easyw/kicadStepUpMod #87: Is there a way to make pads show up on the bottom Layer (B.Cu)?

BillyPocketNC, 2021-06-14:
Is it possible to label pads so that they export to KiCad Footprints on the B.Cu layer?

easyw, 2021-06-20:
@BillyPocketNC 
ATM there is no way to obtain a SMD pad on bottom layer... you need to later edit the pads in kicad fp editor

sslupsky, 2021-09-03:
It would be useful to be able to specify a mid layer as well.

easyw, 2021-09-16:
@sslupsky @BillyPocketNC 
it should now be available:
simply add `_B_Cu_` or `_In1_Cu_` to the Sketch label to set the layer to SMD, Poly or NetTie pads

I have added also the option to assign a pad number (useful in case of single pad each sketch):
simply add `_padNbr=1` or `_padNbr=12` to the Sketch label to set the number of the pads included in the sketch

Please test this new features

luzpaz, 2022-04-30:
@easyw this should be added to documentation and ticket closed?