easyw/kicadStepUpMod #91: Crash exporting footprint

sslupsky, 2021-09-04:
I downloaded the most recent version of ksu today (using the add-on manager) and it is crashing when I attempt to export a footprint.
```
need to discretize Arcs
14:25:27  added circle pad
14:25:27  added discretized polygon
14:25:27  Running the Python command 'ksuToolsFootprintGen' failed:
Traceback (most recent call last):
  File "/Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 2065, in Activated
    kicadStepUptools.PushFootprint()
  File "/Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 16939, in PushFootprint
    from PartGui import BOPTools

cannot import name 'BOPTools' from 'PartGui' (/Applications/FreeCAD 0.20.app/Contents/Resources/lib/PartGui.so)
```

I confirmed BOPTools can be imported using the FreeCAD console:
```
>>> import BOPTools
>>> BOPTools.generalFuseIsAvailable()
True
>>> import Part
>>> Part.OCC_VERSION
'7.5.2'
>>> 
```

FreeCAD version:
```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.25541 (Git)
Build type: Release
Branch: master
Hash: ca043dcb6d20d70b8e24d75e810683214eecc5d4
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: C/Default (C)
```

File used to replicate the issue:
[test.FCStd.zip](https://github.com/easyw/kicadStepUpMod/files/7110477/test.FCStd.zip)

sslupsky, 2021-09-04:
I switched back to commit e726bf177e7243189070c8e5886e837ec006b483 and the export works.

easyw, 2021-09-05:
fixed with https://github.com/easyw/kicadStepUpMod/commit/3ef67ccbd93cd9aab2e1b3b32679d5bc722ac320

easyw, 2021-09-06:
it seems the issue is related to `import BOPTools.JoinFeatures`
It should now be fixed both for https://github.com/easyw/kicadStepUpMod/issues/92 and https://github.com/easyw/kicadStepUpMod/issues/91

sslupsky, 2021-09-07:
Yes, I think this has been resolved with this commit.  Thank you!

closed 2021-09-07 by sslupsky