easyw/kicadStepUpMod #92: Python exception when exporting footprint

sslupsky, 2021-09-05:
Footprint export exception using test file uploaded with issue #91 but with a different exception using the most recent commit 3ef67ccbd93cd9aab2e1b3b32679d5bc722ac320:

```
13:10:47  PoM not present
13:10:47  using 'Part' container and 'Links'
13:10:47  FC Version 020-25541
13:10:47  Asm3 WB present
13:10:47  kicad StepUp version 9.7.5.7
13:10:47  tolerance on vertex applied
13:10:47  applying Materials to Shapes
13:10:47  your home path is /Users/stevenslupsky
13:10:47  export to STEP False
13:10:47  Tabified done !
13:10:47  done!
13:10:47  need to discretize Arcs
13:10:47  added circle pad
13:10:47  added discretized polygon
13:10:47  Running the Python command 'ksuToolsFootprintGen' failed:
Traceback (most recent call last):
  File "/Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 2065, in Activated
    kicadStepUptools.PushFootprint()
  File "/Users/stevenslupsky/Library/Preferences/FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 16954, in PushFootprint
    sv0 = Draft.makeShape2DView(FreeCAD.ActiveDocument.getObject(Connect.Name), FreeCAD.Vector(-0.0, -0.0, 1.0))

Cannot access attribute 'Name' of deleted object
```

sslupsky, 2021-09-05:
I poked around a bit and this commit cc1a14dbe95af94b5e9431ed7e2c99bca9f42721  appears to be the one that broke it.

easyw, 2021-09-05:
@sslupsky 
I would need your full FC info details
I cannot replicate your issue

sslupsky, 2021-09-07:
I am running one of the FreeCAD 0.20 builds.  Here is the FreeCAD info:
```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.25541 (Git)
Build type: Release
Branch: master
Hash: ca043dcb6d20d70b8e24d75e810683214eecc5d4
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: C/Default (C)
```

I also tried with the 0.19 stable version and observe the same error.
```
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)
```

sslupsky, 2021-09-07:
The problem appears to have something to do with the arc at the top of the pad.  If I delete that arc and replace it with a line, I do not observe the exception.

sslupsky, 2021-09-07:
There was some issue with the pull and it was still using 9.7.5.7.  I re-synced the repo and now it is showing version 9.7.5.9 (not sure why it's not 9.7.6.0).  This has resolved the exception.  Thank you!

closed 2021-09-07 by sslupsky