easyw/kicadStepUpMod #93: 'ksuToolsAddTracks' failed

HaydenHu, 2021-09-06:
22:42:35  Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
  File "D:\SOFTWARE\FreeCAD_0_20\Mod\kicadStepUpMod-master\kicadStepUpCMD.py", line 3359, in Activated
    import tracks
  File "D:\SOFTWARE\FreeCAD_0_20\Mod\kicadStepUpMod-master\tracks.py", line 8, in <module>
    import kicad_parser
  File "D:\SOFTWARE\FreeCAD_0_20\Mod\kicadStepUpMod-master\kicad_parser.py", line 8, in <module>
    from future.utils import iteritems

No module named 'future'

easyw, 2021-09-06:
@HaydenHu 
would you please post your full FC info?
It is strange that module 'future' is missing

HaydenHu, 2021-09-06:
> @HaydenHu
> would you please post your full FC info?
> It is strange that module 'future' is missing
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.20.25566 (Git)
Build type: Release
Branch: master
Hash: 8c361dd3fc83840a911da950a845eb9f9342a0b6
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: Chinese/China (zh_CN)

easyw, 2021-09-06:
@HaydenHu 
I should have removed the 'future' dependency ... still it is strange your version doesn't have 'future' module.

easyw, 2021-09-08:
It should be solved... feel free to reopen in case of issues

closed 2021-09-08 by easyw