easyw/kicadStepUpMod #102: "Push 3D moved model(s) to PCB" failing due to timestamp issue

cfunseth, 2021-11-16:
I believe this is different than #75 so I wanted to report it here.

I've created a fairly simple design with seven LEDs on a unique PCB shape.  I'm attempting to align the LEDs with external geometry by hand using the "Transform" tool on each LED.

All goes well for transforming the first six LEDs (D1 through D6) and pushing them back to the board, but on the seventh LED (D7) I get the error "footprint Timestamp not found! Please reload the kicad_pcb file" any time I attempt to push it back to the board.  I've recreated the project twice, verified that I'm using the latest version of the workbench, and have tried using multiple LED styles, but keep running into the same issue.

What's interesting is that if I add an eight "dummy" LED that isn't associated with anything on the schematic, then the problem goes away for the seventh LED and moves to the "dummy" LED any time I try to move that one.  I'm not sure this "fixes" the issue though as I've started noticing some translation issues when I try to sync the 3D positions back to the PCB.

I'm at a bit of a loss and was hoping someone here might have some suggestions or insights into what might be going on.  Please let me know if there's anything I can send to help narrow in on the problem.

easyw, 2021-11-16:
@cfunseth 
would you please post a minimal pcb file that fails?

cfunseth, 2021-11-16:
@easyw 
Here's the file I've been using, I've moved D1 through D6 in this file and D7 is still showing the error when I try to push back to the PCB:
[example.zip](https://github.com/easyw/kicadStepUpMod/files/7547947/example.zip)

cfunseth, 2021-11-17:
This is also the error that pops up:
[image]

cfunseth, 2021-11-17:
I am a novice at Python, but I didn't just want to complain without putting any work in.

I noticed that the timestamp check in the "push3D2pcb" function was changed in the last year from:
`if len(re.findall('\s\(tstamp(\s'+tsp+'.+?)\)',data, re.MULTILINE|re.DOTALL))>0:`
to:
`if len(re.findall('\s\(tstamp(\s.+?'+tsp+'.+?)\)',data, re.MULTILINE|re.DOTALL))>0:`

When I use the previous version of this check, I don't encounter the same error with moving the D7 component and pushing it back to the PCB.  I'm not well versed in regex expressions but wanted to at least do some of the leg work to help in figuring out this issue.

Hope it helps!

easyw, 2021-11-17:
@cfunseth 
it should be fixed now... 
https://github.com/easyw/kicadStepUpMod/commit/f8ea624dfcbd37bea00bb49bb4050a76abc7c941
thanks for your detailed report and suggestions

closed 2021-11-18 by easyw