easyw/kicadStepUpMod #103: Missing faces on 3D file after export for footprint

2EyeTec, 2021-11-20:
I added a Step file to a footprint in Freecad and exported it to the respective Kicad folder. When I view the 3D in Kicad, everything looks good. When I open the PCB in FC the solids are missing some faces.

Take a look at the screenshots: the left body is the original file and the one on the right is the exported version from FC/KicadStepUp. The back nut, the pins and the part where the pins go into the connector are broken.

[image]
[image]

Here are the two step files:

[step.zip](https://github.com/easyw/kicadStepUpMod/files/7574977/step.zip)

easyw, 2021-11-21:
@2EyeTec 
would you please post the FC full version and StepUp version?
I'm on FC0.20, verKSU="10.1.7.8" both on win and Linux and I don't have any issue...
[test-exporting.zip](https://github.com/easyw/kicadStepUpMod/files/7576813/test-exporting.zip)

2EyeTec, 2021-11-22:
It's FC 0.19 and KSU 10.1.7.8

Thanks!

easyw, 2021-11-22:
@2EyeTec 
please post full FC info

I have no issue also on 
```
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0

```

2EyeTec, 2021-11-22:
OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: 0d9536ed3e8c7f40197b5606e1b7873625e1d6fe
Python version: 3.8.10
Qt version: 5.12.8
Coin version: 4.0.0
OCC version: 7.5.2

easyw, 2021-11-22:
@2EyeTec 
this is an issue on your FC/OCC release
no issue on 
```
OS: Linux Mint 20.2 (X-Cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0

```
I think you should stay on a up-to-date AppImage or open an issue at FreeCAD

closed 2021-11-22 by easyw