easyw/kicadStepUpMod #106: add track button error: "list indices must be integers or slices, not str"

smarq8, 2021-12-25:
using "add tracks" fail for one of my board, for another work fine.

freecad:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Polish/Poland (pl_PL)

kicad v5.1.12

```
19:54:29    making 10 lines
19:54:29    checking edge cuts
19:54:29      making 10 lines
19:54:29    making 10 lines
19:54:29    checking edge cuts
19:54:29      making 8 lines
19:54:29      making 4 arcs
19:54:29      making 1 polys
19:54:30  Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
  File "C:\Users\xXx\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUpCMD.py", line 3409, in Activated
    tracks.addtracks()
  File "C:\Users\xXx\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\tracks.py", line 346, in addtracks
    pcb.makePads(shape_type='face',thickness=0.05,holes=True,fit_arcs=True) #,prefix='')
  File "C:\Users\xXx\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 1617, in makePads
    w = self._makeCustomPad(p)
  File "C:\Users\xXx\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 1538, in _makeCustomPad
    wire,width = makePrimitve(key, getattr(params.primitives, key))
  File "C:\Users\xXx\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 322, in makePrimitve
    width = getattr(params,'width',0)
  File "C:\Users\xXx\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\fcad_parser\sexp_parser\sexp_parser.py", line 177, in __getattr__
    return self.__getitem__(name)
  File "C:\Users\xXx\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\fcad_parser\sexp_parser\sexp_parser.py", line 147, in __getitem__
    v = self._value[key]

list indices must be integers or slices, not str
```
[image]
[angleMeter dev board.zip](https://github.com/easyw/kicadStepUpMod/files/7776182/angleMeter.dev.board.zip)

easyw, 2021-12-28:
not all the footprint are fully readable by the tools...
in your case the jumper is the culprit
using a different fp for the jumper is solving your issue
[SolderJumper-2_P1.3mm_Bridged_RoundedPad1.0x1.5mm.zip](https://github.com/easyw/kicadStepUpMod/files/7786596/SolderJumper-2_P1.3mm_Bridged_RoundedPad1.0x1.5mm.zip)

smarq8, 2022-01-08:
Which part of current fp is unreadable? Or how to determine which fp make issue, because from FC log I can not get any clue or I miss something. It might be usuful info in the future.

easyw, 2022-01-12:
> Which part of current fp is unreadable?

```
    (pad 2 smd custom (at 0.65 0) (size 1 0.5) (layers F.Cu F.Mask)
      (net 88 "Net-(JP1-Pad2)") (zone_connect 2)
      (options (clearance outline) (anchor rect))
      (primitives
        (gr_circle (center 0 0.25) (end 0.5 0.25) (width 0))
        (gr_circle (center 0 -0.25) (end 0.5 -0.25) (width 0))
        (gr_poly (pts
           (xy 0 -0.75) (xy -0.5 -0.75) (xy -0.5 0.75) (xy 0 0.75)) (width 0))
      ))
```
the issue is when nceatingnthe pad as sum of circles and a poly... it could have done with a single poly zone or with a roundrect pad plus a rect

> Or how to determine which fp make issue, because from FC log I can not get any clue or I miss something.

check the last footprint created before the error and inside the file find the next one... that is the culprit
in our case the last one is 'R12' (line 3318) ... the following fp inside the file is 'JP1' (line 3354)

h-actsafe, 2022-11-10:
I recently discovered that this issue happens also with the Infineon_PG-HSOF-8-1_ThermalVias footprint from the kicad standard library. The error message is entirely unhelpful in pointing out what the user needs to do to make the export work.

I feels reasonable to assume that footprints that are in the KiCad standard library (and thus, have been properly done), should be possible to use? 

I have created a simple demo project which only includes this footprint, and it imports fine as 3d-model, but importing tracks fails with the traceback below.
[KiCadStepUp Import fail.zip](https://github.com/easyw/kicadStepUpMod/files/9981341/KiCadStepUp.Import.fail.zip)

```
Traceback (most recent call last):
  File "C:\Users\henpro\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicadStepUpCMD.py", line 3805, in Activated
    add_toberemoved = tracks.addtracks()
  File "C:\Users\henpro\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\tracks.py", line 392, in addtracks
    pcb.makePads(shape_type='face',thickness=0.05,holes=True,fit_arcs=True) #,prefix='')
  File "C:\Users\henpro\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 1707, in makePads
    w = self._makeCustomPad(p)
  File "C:\Users\henpro\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 1628, in _makeCustomPad
    wire,width = makePrimitve(key, getattr(params.primitives, key))
  File "C:\Users\henpro\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\kicad_parser.py", line 326, in makePrimitve
    width = getattr(params,'width',0)
  File "C:\Users\henpro\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\fcad_parser\sexp_parser\sexp_parser.py", line 177, in __getattr__
    return self.__getitem__(name)
  File "C:\Users\henpro\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\fcad_parser\sexp_parser\sexp_parser.py", line 147, in __getitem__
    v = self._value[key]

list indices must be integers or slices, not str
```
Build details:
```
14:59:06  FreeCAD build date: 2021/04/15
14:59:06  STEP UseAppPart available
14:59:08  kicad_parser_version 2.1.7
14:59:08  tracks version: 2.5.6
14:59:12  kicad version=20211014
```

[KiCadStepUp Import fail.zip](https://github.com/easyw/kicadStepUpMod/files/9981341/KiCadStepUp.Import.fail.zip)

h-actsafe, 2022-11-10:
Playing a bit more with the footprint it seems to be pads 2 and 3 (not the vias) that are causing issues in this particular footprint. Removing them makes import work.

PR #157 (2023-01-17): Fix tracks import for custom pad with multiple primitives of the same type
Track import would fail with a `list indices must be integers or slices, not str` error if a custom pad used multiple instances of the same primitive type (e.g. multiple gr_arcs), but worked if each type of primitive appears at most once.

The SexpParser is a little weird in that the type of items changes if there's just 1 item vs 2 with the same key, but this seems like a pattern used elsewhere in `kicad_parser.py` to handle both cases.

Fixes #106 and is probably a preferable fix compared to #111

Video demonstrating  [... cut]
closed 2023-01-17 by easyw