easyw/kicadStepUpMod #107: local variable 'curve' referenced before assignment

kishanmbhat, 2021-12-29:
Hi,

Found an unusual bug after the latest version (version 10.1.9) of KiCadStepUp workbench was installed on FreeCAD (version 0.19 build 24291). I am importing kicad_pcb file that is developed on Kicad 6.0.0 and the workbench fails to load the PCB board stating that - **local variable 'curve' referenced before assignment**. Please find the issue log below. The same file has been successfully loaded in the previous versions of the workbench on FreeCAD and also successfully rendered in the 3D viewer of KiCad 5.1.12 and KiCad 6.0.0.

Have attached the kicad_pcb file and the mounting hole footprint used in the PCB design as a zip file.

Issue Log
------------
opening /media/kishan/KISHAN2/Office Backups/Ubiq/Switch/Design/Panelization/Proto_V2.2/Test.kicad_pcb
21:04:19  my file path /media/kishan/KISHAN2/Office Backups/Ubiq/Switch/Design/Panelization/Proto_V2.2
21:04:19  grid origin not set
using default top left corner
21:04:20  PCB Loader 
21:04:20  pcb thickness 1.6mm
21:04:20  kicad_pcb version 20211014
21:04:20  parsing
21:04:20  4 edge segments
21:04:20  aux origin not found
21:04:20  grid origin not set
default value on top left corner
21:04:20  parsing&building time 0.634
21:04:20  Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "/home/kishan/.FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 843, in Activated
    kicadStepUptools.onLoadBoard()
  File "/home/kishan/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 6788, in onLoadBoard
    modules,nsk = DrawPCB(mypcb,SketchLayer,override_pcb,keep_pcb_sketch)
  File "/home/kishan/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 12446, in DrawPCB
    sa,ea = arcAngles2(arc1,curve)

local variable 'curve' referenced before assignment

KiCad version info
----------------
Application: KiCad PCB Editor
Version: 6.0.0-d3dd2cf0fa~116~ubuntu20.04.1, release build
Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.11.0-43-generic x86_64, 64 bit, Little endian, wxGTK, zorin, x11
Build Info:
	Date: Dec 23 2021 14:23:02
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.68.0
	ngspice: 31
	Compiler: GCC 9.3.0 with C++ ABI 1013
Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

FreeCAD version info
--------
OS: Zorin OS 16 (zorin:GNOME/zorin)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/India (en_IN)

[KiCadStepUp_curve_referenced_b4_assignment_Bug.zip](https://github.com/easyw/kicadStepUpMod/files/7789629/KiCadStepUp_curve_referenced_b4_assignment_Bug.zip)

easyw, 2021-12-29:
hi @kishanmbhat 
thanks for the kicad_pcb test file...
it should be fixed with this commit https://github.com/easyw/kicadStepUpMod/commit/be679a89e6fe3bb4f24bad8bd7881191c046a74e

kishanmbhat, 2021-12-29:
hi @easyw,

Thank you very much for the quick turnaround. Works like a charm.

cheers!

> hi @kishanmbhat thanks for the kicad_pcb test file... it should be fixed with this commit [be679a8](https://github.com/easyw/kicadStepUpMod/commit/be679a89e6fe3bb4f24bad8bd7881191c046a74e)

closed 2021-12-29 by kishanmbhat