easyw/kicadStepUpMod #108: Moved 3D parts don't saved correctly.

rahmanshaber, 2021-12-31:
[image]

Here i changed some parts position in freecad and pushed the change using this button and waited to get the message that all change is saved>> then opened the brd file from kicad >> then the 3d view parts position doesn't match. also brd file don't show the same potion that i pushed from freecad.

[image]

You can test it using the raspberry pi CM4 board file from here https://www.raspberrypi.com/products/compute-module-4-io-board/

Using latest kicad 6 and freecad 19.3

easyw, 2022-02-09:
this seems related to the transformation applied to the 3d model to be moved...
it is a good practice to have 3D models with standard placement as in the kicad fp libraries
x,y,z 0,0,0
rotation 0

leventelist, 2022-11-01:
I have similar issue. I moved a component in FreeCAD, and the rotation of the component changed from -90 to 90 degrees.

easyw, 2023-03-30:
I have added a warning in case of x,y rotation in 3D models