easyw/kicadStepUpMod #118: importing pads from board puts them into coordinates relative to the page origin

EeliK, 2022-02-27:
Importing copper with "Add tracks" button imports pads, but puts them into easily recognizable wrong location: their coordinates are relative to the (internal) coordinate origin. This is easy to see if the board edge is a rectangle drawn along the page area borders. The coordinates should be calculated as the footprint location + pad location.

[image]

[image]

[image]

EeliK, 2022-02-27:
Locations are correct if tracks are imported without a pre-loaded board.

easyw, 2022-02-27:
please post a board that is failing

EeliK, 2022-02-27:
[smalltestproject.zip](https://github.com/easyw/kicadStepUpMod/files/8149227/smalltestproject.zip)

easyw, 2022-02-28:
this seems an issue when there is only one components per layer... if you add at least one part each layer it seems it is solved

g3grau, 2026-09-24:
I still had the same offset issue (I didn't try to place a dummy component on the backside). My boards moved the grid origin to the center. The board is properly imported, the traces are imported to the original origin. For kicad 10 to freecad 26.3 I came across some other problems like layer naming/numbering changes (F.Adhes is searched for but it is called Adhesive, and it is not a conductor layer), zones which define multiple layers, and some "bad shapes" etc. Due to lack of time I let codex run an hour across it until my boards completed without issues (experimental fork is on GH).