easyw/kicadStepUpMod #119: Edge.Cuts from PARTS are not taken into account

slaesh, 2022-03-08:
I have two use-cases

1. Parts has Edge.Cuts for mountings or cable-pass-through (not just a whole, actual cut-outs!)
2. using a "part" as the "PCB-template". Only the part has the Edge.Cutouts

Both are not working right now :)

slaesh, 2022-03-10:
maybe we can shortcut here: `should it be possible?`

Guess I have a "non usual" setup..
.. using kicad v6, self-compiled. (arch linux)
.. and the freecad branch from realthunder.

[image]
[kicadstepup-part-cutout-test.zip](https://github.com/easyw/kicadStepUpMod/files/8222444/kicadstepup-part-cutout-test.zip)

easyw, 2022-03-10:
> maybe we can shortcut here: `should it be possible?`
> 
> Guess I have a "non usual" setup.. .. using kicad v6, self-compiled. (arch linux) .. and the freecad branch from realthunder.
 
This issue is related to the use of fp with polyline edge cuts...
ATM only fp with edge cuts segments are handled...
having the test board I will see if I can include your setup in the code...

slaesh, 2022-03-10:
oh, I see! Just changed to multiple lines now instead of the one polyline. Thanks for the hint ;)

But if you find time, would be cool to use whatever kicad offers :)
I will let it open, but feel free to close it =)

Anyway - thanks for your time and this awesome project! :muscle: :star_struck:

easyw, 2022-03-17:
@slaesh 
it should support also polylines in footprints at least for kicad v6 both for pcb and footprint loading in FreeCAD...
kv7 only for pcb loading in FreeCAD ATM...

luzpaz, 2022-05-11:
>  having the test board I will see if I can include your setup in the code...

@slaesh do you mind providing this?

slaesh, 2022-05-11:
hey @luzpaz  :)

did you see this zip attached?

https://github.com/easyw/kicadStepUpMod/issues/119#issuecomment-1063894445

it's quite a few days till then.. so if that is not enough, I may create another demo tomorrow. 👍

closed 2022-10-17 by easyw

luzpaz, 2022-10-19:
@easyw is there a commit associated with this ticket ?