easyw/kicadStepUpMod #121: Footprint generator only works for rectangular and circular sketch elements

alowell-ssl, 2022-03-13:
I first noticed this on realthunder's snap package, but I'm seeing the same issue in the official weekly build of 0.20. I'm using Kicad 6.0.2 on Ubuntu.

In the Pads_TH_SMD sketch, I tried a number of shapes such as polygons with the polyline tool.  When I run the export footprint tool and view the footprint in Kicad, the polygonal pads get mapped to just a rectangular pad.

I noticed that if I draw the pads in FreeCAD sketcher using only rectangle and circle elements, the pads import OK into Kicad.  I also noticed that the example footprint-template.FCStd works OK.  However, if I make some modification to footprint-template.FCStd, then the exporting once again doesn't work correctly.  I tried using some of the other tools in StepUp like "Validate selected sketch", but these don't appear to fix the problems.

I will also note that the PCB edge export tool appears to work fine on complicated shapes.  So the issue seems limited to the footprint export tool.

Maybe I'm not using the tool correctly?

alowell-ssl, 2022-03-13:
OK I think I just wasn't using the tool properly... If I instead use the "Pads_Poly" for the sketch containing the polygons (as well as include a reference pad shape internal to the polygon) then I am able to get the polygon shapes in Kicad.

Is there a reference for how the names of the sketches in FC map to the Kicad layers?  And what names to use for which types of shapes?  I had assumed from the basic example that all I need to do was name my sketch Pads_TH_SMD.

Thanks

easyw, 2022-03-14:
@alowell-ssl 
> Is there a reference for how the names of the sketches in FC map to the Kicad layers?

unfortunately not ATM ... I often suggested to have some user writing some doc (I would help in the process)... but ATM I had very little feedback