easyw/kicadStepUpMod #123: half-hole pad

HaydenHu, 2022-03-18:
When importing a PCB with a half-hole pad, there is a residue on the cutting pad, and there is no copper plating layer in the pad hole, and it is hoped that a copper plating layer is added to the via, just like whether it is possible to generate via, make a parameter setting, set by the user.
[image]

easyw, 2022-03-18:
> When importing a PCB with a half-hole pad, there is a residue on the cutting pad

@HaydenHu 
could you please add a failing board?

> there is no copper plating layer in the pad hole, and it is hoped that a copper plating layer is added to the via, just like whether it is possible to generate via, make a parameter setting, set by the user

for this feature please evaluate [fcad_pcb](https://github.com/realthunder/fcad_pcb)
StepUp is aimed to obtain a mechanical 'image' of the pcb, not a full reproduction

HaydenHu, 2022-03-19:
> 
[NRF24L01.zip](https://github.com/easyw/kicadStepUpMod/files/8308621/NRF24L01.zip)

I wanted to use it to render more realistic models, and fab didn't quite fit

easyw, 2022-03-19:
> When importing a PCB with a half-hole pad, there is a residue on the cutting pad

@HaydenHu 
the cutting pads should be fine with latest release

About the no copper plating layer in the pad holes, this is by design to get the model file size lighter

HaydenHu, 2022-03-19:
@easyw 
I can turn this on with Settings like this one, turn it on when I need it, and generate lighter models when I don't
[image]

easyw, 2022-03-19:
@HaydenHu 
> I can turn this on with Settings like this one, turn it on when I need it, and generate lighter models when I don't

Adding plating copper for drills is generally very intensive both for file size and loading time
ATM there is no way to add plating copper for drills in StepUp

HaydenHu, 2022-03-19:
Copper plating is not added by default and can be turned on by those who need it

HaydenHu, 2022-03-19:
[image]
When eagle's PCB is imported into Fusion360, the solder pad will generate copper plating layer. If only the copper plating layer is added to the through-hole solder pad, it will not affect the generation time. Via can not generate copper plating layer, because the through-hole solder pad is not very fixed.
@easyw

easyw, 2022-03-19:
@HaydenHu 
>    If only the copper plating layer is added to the through-hole solder pad, it should not affect the generation time. 

what does let you think so?

>    There are not many through-hole pads

It depends on the user's board

Anyway ATM, as I already said, there is no easy way to add copper plating to the pads with StepUp... if you are willing to make a Pull Request, I will be happy to evaluate if it could be merged to main code.

HaydenHu, 2022-03-19:
[image]
[image]
This PCB is the correct generation of one side
[MRR_VR.zip](https://github.com/easyw/kicadStepUpMod/files/8309672/MRR_VR.zip)

easyw, 2022-03-19:
@HaydenHu 
it should be fixed now

HaydenHu, 2022-03-20:
If the half-hole footprint is rotated 180 degrees, the pad will be missing
[image]

[MRR_VR.zip](https://github.com/easyw/kicadStepUpMod/files/8310931/MRR_VR.zip)

easyw, 2022-03-20:
@HaydenHu 
thanks for reporting it... StepUp is aimed mostly to a mechanical interoperability and it covers a subset of kicad formats...

> I wanted to use it to render more realistic models

If this is your need, I would suggest you to move to Blender...
https://hackaday.io/project/175077-remoticon-kicad-to-blender-pcb-renders
https://forum.kicad.info/t/export-pcb-devices-with-textures-for-rendering-in-blender-etc/7490
https://www.youtube.com/watch?v=RcMIcRqMWn8
and here for even 3D solder 
[image]

https://blog.pcb-arts.com/en/blog/blender-tutorial-1
https://blog.pcb-arts.com/en/blog/blender-tutorial-2
https://blog.pcb-arts.com/en/blog/blender-tutorial-3

closed 2022-03-20 by easyw