easyw/kicadStepUpMod #127: Possible regression: Trapezoid pads not imported correctly into FreeCAD.

der-ule, 2022-03-25:
Hello @easyw, I was playing with an old board and after trying to import it in FreeCAD, I noticed that the issued that you fixed last time, has come up again (the slider is generated with the integrated KiCad scripts):

https://forum.kicad.info/t/stepup-support-for-trapezoids-pads/20978

KiCad:
[image]

FreeCAD:
[TouchSlider-5_17x3.5mm.zip](https://github.com/easyw/kicadStepUpMod/files/8350276/TouchSlider-5_17x3.5mm.zip)

[image]

KSU:
```
12:57:57  PoM not present
12:57:57  using 'Part' container and 'Links'
12:57:57  kicad_parser_version 2.1.3
12:57:57  FC Version 019-24276
12:57:57  kicad StepUp version 10.4.0
12:57:57  tolerance on vertex applied
12:57:57  applying Materials to Shapes
12:57:57  your home path is C:\Users\uestrada
12:57:57  export to STEP False
12:57:57  Tabified done !
12:57:57  done!
12:57:57  PoM not present
12:57:57  using 'Part' container and 'Links'
12:57:57  FC Version 019-24276
12:57:57  kicad StepUp version 10.4.0
12:57:57  tolerance on vertex applied
12:57:57  applying Materials to Shapes
12:57:57  your home path is C:\Users\uestrada
12:57:57  kicadStepUp
12:57:57  closed
12:57:57  KISYS3DMOD=/
module_3D_dir=//../
12:58:07  FC Version 019
12:58:07  VBO status False
12:58:07  FootPrint Loader D:/Data/Documents/100_KiCAD/user6_lib/footprints/Tests.pretty/TouchSlider-5_17x3.5mm.kicad_mod
12:58:07  opening Transaction 'opening_kicad_footprint'
12:58:07  TouchSlider_5_17x3_5mm_fp
12:58:07  closing Transaction 'opening_kicad_footprint'
```

KiCad Version:
```
Application: KiCad PCB Editor (64-bit)

Version: (6.0.4-53-gbf8c7d7697), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 18362), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 24 2022 22:10:09
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.6.0
	Curl: 7.78.0-DEV
	ngspice: 36
	Compiler: Visual C++ 1928 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

FreeCAD version:
```
OS: Windows 10 Version 1903
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)

```

easyw, 2022-03-25:
@der-ule 
I'm not sure if the fix was for track importing or fp importing...
Loading tracks from kicad_pcb is still working (loading trapezoid correctly)
[image]

Loading fp from kicad_mod is not working, but I'm not sure if ever worked

der-ule, 2022-03-28:
You are totally right, I don't remember ever trying with the footprint importer, I just wasn't aware that they were two different code bases.

I imported the board and besides a minor glitch, it imported correctly:

[image]

Thanks a lot !

closed 2022-04-04 by easyw