easyw/kicadStepUpMod #128: Can not find custom 3d model path in FreeCAD

bookshiyi, 2022-04-19:
Hello,
System: Manjaro latest version
KiCad : 6.0.4
FreeCAD: 0.19
kicadStepUpMod: latest version

When I load pcb from KiCad to FreeCAD, I received some error message about kicadstepupmod can not find 3d model which in my user path(not  KiCad system path).
 
```
 missing module(s)
... searching path:
/usr/share/kicad/3dmodels/
/home/`USER`/repos/kicad_library/3dmodel/
... missing module(s) '.step' or '.stp' or .iges' or '.igs'
"${KICAD6_USER_3DMODEL_DIR}/TF-SMD_L14.8-W14.5-H1.9-P1.1.step" (.stp or .step)
"${KICAD6_USER_3DMODEL_DIR}/TF-SMD_L14.8-W14.5-H1.9-P1.1.wrl" (.stp or .step)

. . . missing 10 model(s)
```

I have configured `KICAD6_USER_3DMODEL_DIR` both in KiCad and FreeCAD， like this:

In KiCad
[image]

In FreeCAD
[image]

Thanks.

easyw, 2022-04-20:
@bookshiyi 
from your log  it seems there is something strange in the secondary 3D folder location:

```
/home/`USER`/repos/kicad_library/3dmodel/
```
it seems there is a  backtick character before U and before / in your secondary location string...

bookshiyi, 2022-04-20:
> @bookshiyi from your log it seems there is something strange in the secondary 3D folder location:
> 
> ```
> /home/`USER`/repos/kicad_library/3dmodel/
> ```
> 
> it seems there is a backtick character before U and before / in your secondary location string...

Sorry, I just want to mask the username.
In the actual log from kSU, it is indeed displayed the path I filled in FreeCAD settings.

easyw, 2022-04-21:
but do you have i.e. ths model in your system?

/home/USER/repos/kicad_library/3dmodel/TF-SMD_L14.8-W14.5-H1.9-P1.1.step
?
Which is your USER correspondent in your case?

${KICAD6_USER_3DMODEL_DIR}/TF-SMD_L14.8-W14.5-H1.9-P1.1.step

bookshiyi, 2022-04-22:
Hi,

I indeed have the 3d model `TF-SMD_L14.8-W14.5-H1.9-P1.1.step`  in  the path `/home/USER/repos/kicad_library/3dmodel/TF-SMD_L14.8-W14.5-H1.9-P1.1.step`.

I can find it  both path `/home/USER/repos/kicad_library/3dmodel/TF-SMD_L14.8-W14.5-H1.9-P1.1.step` and `${KICAD6_USER_3DMODEL_DIR}/TF-SMD_L14.8-W14.5-H1.9-P1.1.step` in KiCad.

My custom 3d model only not work with kicadStepUp in FreeCAD.

Thanks.
BR

easyw, 2022-04-29:
@bookshiyi 
would you mind to attach a minimal kicad_pcb sample file (zipped) with a failing 3d loading file?

easyw, 2022-05-04:
@bookshiyi 
I'm closing it. If you need more assistance, please feel free to reopen it.

closed 2022-05-04 by easyw