easyw/kicadStepUpMod #130: Running the Python command 'ksuToolsOpenBoard' failed

yueduz, 2022-04-30:
```
opening /home/lei/kicad/吹奏乐器/吹奏乐器.kicad_pcb
09:13:23  my file path /home/lei/kicad/吹奏乐器
09:13:23  opening Transaction 'opening_kicad'
09:13:23  grid origin found @ (-140, 50)
09:13:23  PCB Loader 
09:13:23  pcb thickness 1.6mm
09:13:23  kicad_pcb version 20211014
09:13:23  parsing
09:13:23  4 edge segments
09:13:23  aux_axis_origin[140, 50]
09:13:23  grid_origin[140, 50]
09:13:23  parsing&building time 0.002
09:13:23  Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "/home/lei/.FreeCAD/Mod/kicadStepUpMod/kicadStepUpCMD.py", line 927, in Activated
    kicadStepUptools.onLoadBoard()
  File "/home/lei/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 6810, in onLoadBoard
    modules,nsk = DrawPCB(mypcb,SketchLayer,override_pcb,keep_pcb_sketch)
  File "/home/lei/.FreeCAD/Mod/kicadStepUpMod/kicadStepUptools.py", line 12393, in DrawPCB
    if p.drill[0] >= min_drill_size or p.drill[1] >= min_drill_size:
  File "/home/lei/.FreeCAD/Mod/kicadStepUpMod/fcad_parser/sexp_parser/sexp_parser.py", line 147, in __getitem__
    v = self._value[key]

(0,)
```

easyw, 2022-04-30:
hi @yueduz 
the issue is in parsing the pad 1 of BT1 fp
The pad has a drill offset but without a drill, which is not (ATM) handled by the ksu parser
I see if I can patch this for the pdb loader... for tracks loader it would take a bit longer
here the pcb with the pad 'reworked'
[吹奏乐器-no-offset.zip](https://github.com/easyw/kicadStepUpMod/files/8596581/-no-offset.zip)

yueduz, 2022-04-30:
It worked. Thank you very much

closed 2022-04-30 by yueduz