easyw/kicadStepUpMod #131: Q: material properties of FreeCAD not fully exported to KiCad *.wrl model ?

analoghifi, 2022-05-09:
not sure if it's my error or if it's meant to be that way but:
I generate a model in FC an add material properties via "Appearence" [Ctrl+D] 
different appearance for different parts of the model.
for example this way I added (in FC) "gold", "shiny plastic" and "pewter"
as a sub-example: "gold" has a full subset of attributes like "shininess", "diffuse color", "specular color", and so on ...

when I export this part via kicadStepUp (the checkbox "use wrl material" is checked),
and I then answer the colour/material substitution questions dialog of the kicadStepUp exporter all with "as is",

-> so I would expect then, that within the exported *.wrl model the "shininess" and "specular color" would show up as well 
like they were set in FC before ?

but the whole exported model has "diffuseColor" properties only -> no "shininess" and no "specular color"
(--> see attachment)
why is that ?
do I always have to use one of the 36 predefinded colour/material combinations of the kicadStepUp-export dialog ?
I would expect, that the full subset of attributes like "shininess", "diffuse color", "specular color", and so on ... that I can preset in FC are exported 1:1 to the kicad wrl-model

Or is KiCad itself only able to show this 36 predefinded colour/material combinations within its 3D viewer and nothing more ?

[CP_Radial_Black_D8.0mm_P3.50mm.wrl.txt](https://github.com/easyw/kicadStepUpMod/files/8652888/CP_Radial_Black_D8.0mm_P3.50mm.wrl.txt)

easyw, 2022-05-11:
@analoghifi 
the colors are assigned referred to an internal table which has colors as per KiCAD recommendations ...
https://forum.kicad.info/t/tutorial-and-guidelines-for-better-materials-on-3d-models/2321

please  use one of the 36 predefinded colour/material combinations of the kicadStepUp-export dialog

analoghifi, 2022-05-11:
Thanks for your answer,
I found this as well:
-> https://gitlab.com/kicad/libraries/kicad-packages3D/-/raw/master/Vrml_materials_doc/KiCad_3D-Viewer_component-materials-reference-list_MarioLuzeiro.pdf?inline=false
and I wonder how to choose "CAP-RADIAL-BLUE-01" from the kicadStepUp-export dialog -> is this missing / or just named different ?
-------------
beside of that:
I think this is IMHO a kind of useless limitation from KiCad's early days (6 years ago).
If I export the wrl directly from FC and then transfer some material properties with a text editor into my original wrl that was created (scaled and positioned) with kicadStepUp (see attachment above),
I can have any colour and shininess that I want -> the KiCad [6.x] 3D-viewer is fine with that and displays all correctly.

easyw, 2022-05-13:
@analoghifi 
you may be interested on this FC forum topic
https://forum.freecadweb.org/viewtopic.php?p=594947#p594947

> Finally I found this script "PartToVRMLsbc.FCMacro" to export to WRL -> https://goneill.co.nz/freecad.php
> it works perfect for me: scaling is done and all of my "appearance" setting I did in FreeCAD are still present in the exported WRL