easyw/kicadStepUpMod #136: Wrong KICAD6_3DMODEL_DIR

leoheck, 2022-06-30:
Hey @easyw 

I don't use this import board function. But I wanted to have a improved version of it and saw that 3d models from the standard Kicad 5 folder are not being loaded, only my custom components.

[image]

Now I saw this. For some reason, the first field here was set wth the KICAD 5 path. 
[image]

However, maybe the best approach is to have an individual field for each one of the available env variables.
so, in this case, since the files have this KICAD6_3DMODEL_DIR, it would fine the correct value, or nothing. But instead, it found the incorrect value. But if I have a project that is using Kicad 5. The opposite will happen if I change this now. 

The right value on my system (Ubuntu 22.04) for `KICAD6_3DMODEL_DIR` is `/usr/share/kicad/3dmodels/`

easyw, 2022-07-01:
hi @leoheck 

> I don't use this import board function.

wrong choice ;)

> But I wanted to have a improved version of it and saw that 3d models from the standard Kicad 5 folder are not being loaded, only my custom components.

your configuration comes from the installation of kSU which was installed when kv5 was the stable release...
now kv6 is the actual stable; if you make a clean FC and kSU install you will get the standard kv6 3D models path filled, unless your main path is already filled (as in your case)
In general, when more then 10 models are missing in loading it will prompt an info message remembering you to configure your searching paths.
Automatic changing a previous searching path configuration in automatic way is not suggested.
Just reading kSU cheatsheet would have helped in solving this issue.

leoheck, 2022-07-01:
> wrong choice ;)
there arre many benefit in using kSU import instead of k export.

Yeah, but I am usually satisfied with the exported step from Kicad, to design my cases.

> if you make a clean FC and kSU install you will get the standard kv6 3D models path filled

Gotcha.

easyw, 2022-07-01:
> Yeah, then I will keep using .step exported from Kicad since time is precious and we cannot buy it.

sometimes the time you invest in learning a new tool would save time later ;)
In mechanical collaboration, StepUp is highly recommended IMO

closed 2022-07-02 by easyw