easyw/kicadStepUpMod #137: Castelated vias

leoheck, 2022-06-30:
Hello easyw!

Does KicadStepUp handle castellated vias?
I was trying to import tracks. And then got this.

[image]

this is the result, I dot have wires connected to the vias
[image]

leoheck, 2022-06-30:
Actually, it imports vias fine. But in the end, something happens that it cannot add. Maybe it is not being able to cut them in half.

leoheck, 2022-07-01:
Ah, I used the wrong word. I am talking about pads, not vias, sorry.

Here in this design, I am using a standard connector to make these castellated holes. However, just one of those pads have the castellated propertie enabled since Kicad is not fully working/using them. Ony the first castellated holes hax this property. but it is not keeping the castellated pad there.

Do you have any advice for me to check or do to make this work?

easyw, 2022-07-01:
> Do you have any advice for me to check or do to make this work?

if you post a failing example I could have a look at...
it simply should work (at least it worked)
https://github.com/easyw/kicadStepUpMod/issues/123

closed 2022-07-03 by easyw