easyw/kicadStepUpMod #138: Load board does not respect the 3D model visibility

leoheck, 2022-07-07:
It is importing all the modules, even the ones that are marked to not show.
I believe this is ok since we are going to have the 3d model of the part that is hidden in the 3d model.
But it could be set to be not visible on Freecad Too

[image]

2 boards are overlapped here, one should be hidden by default.
[image]

leoheck, 2022-07-07:
This is the expected result
[image]

The model can be added, but hidden as shown here.
[image]

easyw, 2022-07-07:
which is your kicad pcb version? and kSU version?
this feature should be supported but it is available only in kv6 or above and only on recent kSU releases

leoheck, 2022-07-07:
Sorry, forgot this info, so here they are.

Kicad version
```
Application: KiCad

Version: 6.0.6-3a73a75311~116~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.12

Platform: Linux 5.15.0-40-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Jun 20 2022 15:50:05
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.83.1
	ngspice: 36
	Compiler: GCC 11.2.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

I thought my FreeCad was the stable 0.20, but it happens is that it is not. So this may be the issue here. 
```
OS: Ubuntu 22.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.
Build type: Release
Branch: unknown
Hash: 91f84184bf4712ba01234b77dc906c53f76c063d
Python 3.10.4, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.56a
  * Curves 0.5.0
  * fasteners 0.3.46
  * Assembly4 0.12.1
  * kicadStepUpMod 10.13.3
  * ExplodedAssembly
```

leoheck, 2022-07-07:
Ah, the PCB was started with an older version of Kicad a long time ago. And then now, it was ported to Kicad 6. Perhaps this can be related to some old string strings inside the .kicad_pcb.

leoheck, 2022-07-08:
Actually, FreeCad version comes with Addons versions too. Take a look. But it was my mistake that I did not have highlighted it. Is this what you are asking, right?

[image]

easyw, 2022-07-09:
ok, I didn't noticed... so you are on latest kSU... it should work then...
I tried with a kv5 pcb saved in kv6 and it worked....
inside kv6 you should have 'hide' in the 3D model line:

>     (model "${KISYS3DMOD}/Connector_PinHeader_2.54mm.3dshapes/PinHeader_1x03_P2.54mm_Vertical.wrl" hide
>           (offset (xyz 0 0 0))
>           (scale (xyz 1 1 1))
>           (rotate (xyz 0 0 0))
>         )

leoheck, 2022-07-09:
Hey, there is this hide inside as your test-hide-kv6.zip
But it happens that it does not hide
See, this is your test. 
Kicad 
[image]
Freecad, loading the kicad_pcb
[image]

easyw, 2022-07-09:
> But it happens that it does not hide
> See, this is your test.

if you open the test file in kicad, it doesn't show any 3d model because they point to kv5 old 3d path
But in FreeCAD, StepUp is searching in what you have configured as 3d path and then it shows a 3d model...
Still two models are loaded but only one is visible...
This is correct because in pcbnew, as in the following screenshot, two 3d models are assigned to the fp, and only the one with 5pins is set to be shown, the one of 3pins is hidden...
[image]

So StepUp is doing it correctly... moreover it should prompt a message warning of the hidden models...
if you open the main container in FC you will see its content showing two models, the on with 3 pins hidden and the one with 5pins displayed
[image]

easyw, 2022-07-09:
and from your screenshot, simply reading the log on the right, you can read that the tool is loading the 3pins model and the it is hiding it, then it will load the 5 pins model, not touching its visibility

leoheck, 2022-07-09:
I am not understanding. 

I am reporting, that, KSU is loading a 3d model that is *hidden* in Kicad v6. But it is not hiding it. 
It is nice that KSU is importing a hidden model, but it should (or it could) be hidden by default.

The filename of the package you shared says [test-hide-kv6.zip](https://github.com/easyw/kicadStepUpMod/files/9076198/test-hide-kv6.zip). As in "Kicad v6". Then you are showing something misconfigured with Kicad 5 environment variables. 
This is kind of confusing. But then I understand Kicad is not showing 3d models that don't have the right path.

I am not using Kicad 5. I am using Kicad 6.

But in the end, I am reporting that KSU could hide, by default, models that are hidden in Kicad. That's it.

leoheck, 2022-07-09:
Maybe you have a different version of KSU.

easyw, 2022-07-10:
> I am reporting, that, KSU is loading a 3d model that is _hidden_ in Kicad v6. But it is not hiding it.

that is a wrong assumption

> But in the end, I am reporting that KSU could hide, by default, models that are hidden in Kicad. That's it.

that is exactly what the test example is doing...
You need to re-read my previous post ... 

Ok, I'm attaching a zipped test with 3D models on local path... 
there are both kicad nd freecad file.
This will show what you are missing
[test-kv6-prj-files.zip](https://github.com/easyw/kicadStepUpMod/files/9078594/test-kv6-prj-files.zip)

leoheck, 2022-07-10:
> that is a wrong assumption

This is not an assumption. I am reporting that. This is what happens on my system. It is not an assumption. It is a fact and the objective of the report. 

Have you tried to load your project with KSU? The report here is about the issues on KSU and no on Kicad. 

Anyway, this is what I see when I import the new test-kv6-prj-files on FreeCad

I could not even try to check the issue related with the report

[image]

Report view
[image]

closed 2022-07-10 by leoheck