easyw/kicadStepUpMod #139: Cannot import tracks... maybe due to custom names

leoheck, 2022-07-07:
It could not import tracks.

It looks like it could not find cooper layers.

```
21:58:11  tracks version: 2.4.6
21:58:17  cannot find kicad_common
21:58:17  Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
  File "/home/lheck/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 3778, in Activated
    add_toberemoved = tracks.addtracks()
  File "/home/lheck/.local/share/FreeCAD/Mod/kicadStepUpMod/./tracks.py", line 310, in addtracks
    pcb = kicad_parser.KicadFcad(filename)
  File "/home/lheck/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicad_parser.py", line 561, in __init__
    self._initStackUp()
  File "/home/lheck/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicad_parser.py", line 659, in _initStackUp
    self._log('stackup info ignored because copper layer {} is not found',
  File "/home/lheck/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicad_parser.py", line 769, in _log
    getattr(logger,level)('{}{}'.format(self.prefix,msg.format(*arg)))

Replacement index 0 out of range for positional args tuple
```

These are my cooper layers, they don't have standard names, maybe this is related.
```
F.cu is labeled Top.cu
B.cu is labeled Bot.cu
```
[image]

They are being rename in the Board Setup
[image]

But this is easily solved since you can still get the original names (thinking here if you are using python module of the pcbnew)

easyw, 2022-07-07:
here again: could you please add your kicad full version?
have you a minimal pcb sample failing in loading the tracks, to be uploaded here to test?

leoheck, 2022-07-07:
Sure, I forgot that eventyallly, since I am not just working on a single thing, as people use to think, but sorry one more time.
 
Pasting it here from my other reply it the other issue.

Sorry, forgot this info, so here they are.

Kicad version
```
Application: KiCad

Version: 6.0.6-3a73a75311~116~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.12

Platform: Linux 5.15.0-40-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Jun 20 2022 15:50:05
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.83.1
	ngspice: 36
	Compiler: GCC 11.2.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

I thought my FreeCad was the stable 0.20, but it happens is that it is not. So this may be the issue here. 
```
OS: Ubuntu 22.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.
Build type: Release
Branch: unknown
Hash: 91f84184bf4712ba01234b77dc906c53f76c063d
Python 3.10.4, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.56a
  * Curves 0.5.0
  * fasteners 0.3.46
  * Assembly4 0.12.1
  * kicadStepUpMod 10.13.3
  * ExplodedAssembly
```

leoheck, 2022-07-07:
> have you a minimal pcb sample failing in loading the tracks, to be uploaded here to test?
Sure, I will check if I can generate something to share.

> moreover from your log, you have a 'strange' error message:
I've noticed it too. But I am not sure what it means since I thought it was related to KSU. But since you are mentioning it, then maybe it is something that I have to track from where it comes.

easyw, 2022-07-08:
it could be related to the pcb stack ...
`File "/home/lheck/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicad_parser.py", line 659, in _initStackUp
    self._log('stackup info ignored because copper layer {} is not found',`
if you could open your pcb file with a text editor (i.e. geany or similar) and post here the stack content, that could help in debugging
also layers content could help (just before stackup)

> (setup
    **(stackup**
      (layer "F.SilkS" (type "Top Silk Screen"))
      (layer "F.Paste" (type "Top Solder Paste"))
      (layer "F.Mask" (type "Top Solder Mask") (thickness 0.01))
      (layer "F.Cu" (type "copper") (thickness 0.035))
      (layer "dielectric 1" (type "core") (thickness 1.51) (material "FR4") (epsilon_r 4.5) (loss_tangent 0.02))
      (layer "B.Cu" (type "copper") (thickness 0.035))
      (layer "B.Mask" (type "Bottom Solder Mask") (thickness 0.01))
      (layer "B.Paste" (type "Bottom Solder Paste"))
      (layer "B.SilkS" (type "Bottom Silk Screen"))
      (copper_finish "None")
      (dielectric_constraints no)
    **)**

closed 2022-07-19 by easyw

leoheck, 2022-07-19:
I thought I had shared the PCB stack up here. Anyways, here it is.

```sexp
  (layers
    (0 "F.Cu" signal)
    (1 "In1.Cu" power "Gnd1.Cu")
    (2 "In2.Cu" signal "Sig1.Cu")
    (3 "In3.Cu" signal "Pwr.Cu")
    (4 "In4.Cu" power "Gnd2.Cu")
    (31 "B.Cu" signal)
    (32 "B.Adhes" user "B.Adhesive")
    (33 "F.Adhes" user "F.Adhesive")
    (34 "B.Paste" user)
    (35 "F.Paste" user)
    (36 "B.SilkS" user "B.Silkscreen")
    (37 "F.SilkS" user "F.Silkscreen")
    (38 "B.Mask" user)
    (39 "F.Mask" user)
    (40 "Dwgs.User" user "User.Drawings")
    (41 "Cmts.User" user "User.Comments")
    (42 "Eco1.User" user "User.Eco1")
    (43 "Eco2.User" user "User.Eco2")
    (44 "Edge.Cuts" user)
    (45 "Margin" user)
    (46 "B.CrtYd" user "B.Courtyard")
    (47 "F.CrtYd" user "F.Courtyard")
    (48 "B.Fab" user)
    (49 "F.Fab" user)
  )

  (setup
    (stackup
      (layer "F.SilkS" (type "Top Silk Screen") (color "White") (material "Liquid Photo"))
      (layer "F.Paste" (type "Top Solder Paste"))
      (layer "F.Mask" (type "Top Solder Mask") (color "Green") (thickness 0.02) (material "Epoxy") (epsilon_r 3.3) (loss_tangent 0))
      (layer "F.Cu" (type "copper") (thickness 0.035))
      (layer "dielectric 1" (type "prepreg") (thickness 0.086 locked) (material "FR4") (epsilon_r 3.9) (loss_tangent 0.035))
      (layer "In1.Cu" (type "copper") (thickness 0.035))
      (layer "dielectric 2" (type "core") (thickness 0.53 locked) (material "FR4") (epsilon_r 4.25) (loss_tangent 0.035))
      (layer "In2.Cu" (type "copper") (thickness 0.035))
      (layer "dielectric 3" (type "prepreg") (thickness 0.12 locked) (material "FR4") (epsilon_r 4.4) (loss_tangent 0.035))
      (layer "In3.Cu" (type "copper") (thickness 0.035))
      (layer "dielectric 4" (type "core") (thickness 0.53 locked) (material "FR4") (epsilon_r 4.25) (loss_tangent 0.035))
      (layer "In4.Cu" (type "copper") (thickness 0.035))
      (layer "dielectric 5" (type "prepreg") (thickness 0.086 locked) (material "FR4") (epsilon_r 3.9) (loss_tangent 0.035))
      (layer "B.Cu" (type "copper") (thickness 0.035))
      (layer "B.Mask" (type "Bottom Solder Mask") (color "Green") (thickness 0.02) (material "Epoxy") (epsilon_r 3.3) (loss_tangent 0))
      (layer "B.Paste" (type "Bottom Solder Paste"))
      (layer "B.SilkS" (type "Bottom Silk Screen") (color "White") (material "Liquid Photo"))
      (copper_finish "ENIG")
      (dielectric_constraints no)
    )
```