easyw/kicadStepUpMod #141: Traceback while exporting 3D model

Mimosa-oceanis, 2022-07-21:
I've got the STEP file from the manufacturer and I can't export the wrl. I'got this error: 

`09:50:36  Running the Python command 'ksuToolsExportModel' failed:
Traceback (most recent call last):
  File "/home/mimosa/snap/freecad/common/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 1007, in Activated
    routineScaleVRML()
  File "/home/mimosa/snap/freecad/common/Mod/kicadStepUpMod/./kicadStepUptools.py", line 7265, in routineScaleVRML
    lbl=go_export(fullFilePathName)
  File "/home/mimosa/snap/freecad/common/Mod/kicadStepUpMod/./kicadStepUptools.py", line 2657, in go_export
    export(objs, fPathName, 0.3937, lbl)
  File "/home/mimosa/snap/freecad/common/Mod/kicadStepUpMod/./kicadStepUptools.py", line 2277, in export
    color.append(FreeCADGui.ActiveDocument.getObject(obj.Name).ShapeColor)

'Gui.ViewProviderDocumentObject' object has no attribute 'ShapeColor'`

Here are the file so you can test as well. 
[ESP32-S3-WROOM-1.zip](https://github.com/easyw/kicadStepUpMod/files/9157380/ESP32-S3-WROOM-1.zip)

easyw, 2022-07-21:
hi @Mimosa-oceanis 
your model has a lot of component inside the shield... in particular you have a nested of Part containers inside the main Part container... this is not supported ATM
I had to simplify your model before exporting it:
1) I deleted all objects internal to the shield (they are not useful in 3D mechanical integration
2) I made a union of all the objects using the kSU Union button
3) I exported the resulting shape
PS I have added an internal cube to make the union of all shapes as a single structure (fusing the shield with the main pcb)
Attached the modified model
[ESP32-S3-WROOM-simplified-3d.zip](https://github.com/easyw/kicadStepUpMod/files/9159809/ESP32-S3-WROOM-simplified-3d.zip)

Mimosa-oceanis, 2022-07-22:
Hi @easyw 
I understand, thanks a lot.

Mimosa-oceanis, 2022-07-22:
I think I have a similar issue while importing the board.

``opening /home/mimosa/git/Octolink/Octo-one/octo-one.kicad_pcb
11:36:12  my file path /home/mimosa/git/Octolink/Octo-one
11:36:12  opening Transaction 'opening_kicad'
11:36:12  grid origin not set
using default top left corner
11:36:12  PCB Loader 
11:36:12  pcb thickness 1.21mm
11:36:12  kicad_pcb version 20211014
11:36:12  parsing
11:36:12  0 edge segments
11:36:12  aux origin not found
11:36:12  grid origin not set
default value on top left corner
11:36:12  2d circle closed path
11:36:12  parsing&building time 0.002
11:36:12  Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "/home/mimosa/snap/freecad/common/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 927, in Activated
    kicadStepUptools.onLoadBoard()
  File "/home/mimosa/snap/freecad/common/Mod/kicadStepUpMod/./kicadStepUptools.py", line 6606, in onLoadBoard
    modules,nsk = DrawPCB(mypcb,SketchLayer,override_pcb,keep_pcb_sketch)
  File "/home/mimosa/snap/freecad/common/Mod/kicadStepUpMod/./kicadStepUptools.py", line 12096, in DrawPCB
    if md[1] == 'hide':
  File "/home/mimosa/snap/freecad/common/Mod/kicadStepUpMod/./fcad_parser/sexp_parser/sexp_parser.py", line 147, in __getitem__
    v = self._value[key]

(1,)

easyw, 2022-07-22:
@Mimosa-oceanis 
would you mind to send me the board? (if you want in PM at kicad.info forum?  (search for @maui there)

Mimosa-oceanis, 2022-07-22:
I tried to PM in kicad forum, but I can't find a way to start a conversation. Never found the button for PM. here is mine: 
Mimosa-oceanis

easyw, 2022-07-22:
@Mimosa-oceanis 
I have sent you a message at the forum

easyw, 2022-07-22:
@Mimosa-oceanis 
I've pushed an update that should solve your issue...
could you please test it?

Mimosa-oceanis, 2022-07-22:
@easyw 
I'm impressed by your reactivity. Thanks a lot. 
I don't have the error anyway, but I got "FreeCad not responding" window for ages. And it seems the model is too heavy for FreeCAD. I'll wait a bit more to see if Freecad finaly manage to open the board.

easyw, 2022-07-22:
> but I got "FreeCad not responding" window for ages.

could you please post your FC release?
I looked inside kicad_pcb file and I found for some 3D model a path with ':'
i.e.:
`:Octolink_3D_model:Octolink_misc.3dshape/VEL13353R8257G.wrl` 
in which way you added this kind of path?

Mimosa-oceanis, 2022-07-22:
Yes it a second path for my own libraries. 
[explanation: ](https://forum.kicad.info/t/help-understanding-3d-search-paths-and-environment-variables/20726)

The version of freecad I'm using is : 

```
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) Snap 120
Build type: Release
Branch: tag: 0.20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.13.9

```

easyw, 2022-07-22:
@Mimosa-oceanis 
thanks for the link on 'Alias' ... I think I should already cover it
@ line 4332 of  kicadStepUptools.py
`        if (step_module.startswith(':')) or (step_module.startswith('":')):  #alias 3D path`

did your board finished to loading?
Could you share it with all 3d models?
you may use this plugin:
https://github.com/MitjaNemec/Archive3DModels
I use StepUp with big pcb and hundred of big models and it loads quite fast (minutes at most) ...

Mimosa-oceanis, 2022-07-22:
@easyw 
Yes everything was ok. It took a very long time but it managed to load everything. I think all the components are loaded. There juste one mistake on one component who is scaled down on kicad, and in free cad it appears too big. I think this is the only problem. 
I will try using the plugin. 
I managed to export the STEP. you car have it here: [STEP](https://belenos.synology.me:5001/d/s/pbhS4r1WPjgISjBJZnkG9r0dDiP16tL3/-UCN5N-08uvTJUQaIF_C_9axLFE2NX7y-pLmgmIwHsgk)

Thanks a lot for your help.

easyw, 2022-07-22:
@Mimosa-oceanis 
it seems you have some model to be simplified...
I will update later on

easyw, 2022-07-22:
@Mimosa-oceanis 
from your step file, I can see that your U6 and U12 models are container for thousand sub models...
for a mechanical purpose you could simplify those, limiting the model to the pcb and the shield
Tat would help a lot to have a mechanical model to be able to work with handy

easyw, 2022-07-23:
I did a test with CADAssistant 
https://www.opencascade.com/products/cad-assistant/
with this sw of OCC the loading time is less than a 1/10 of FC loading time on the same pc...
there is a lot of room for improvement for FC STEP importing routines

Mimosa-oceanis, 2022-07-27:
@easyw thank you for your help. I will review the parts in order to make them lighter.