easyw/kicadStepUpMod #143: Problem loading board with no holes

cfangmeier, 2022-07-27:
I have a fairly simple board that happens to have no holes in it. The board itself import into FreeCAD ok, but when I try to import the traces with the "ksu tools Add Tracks" button, I get the following error:

```
14:10:20  Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
  File "C:\Users\Caleb\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 3778, in Activated
    add_toberemoved = tracks.addtracks()
  File "C:\Users\Caleb\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\tracks.py", line 411, in addtracks
    extrude_holes(holes,pcbThickness*3)
  File "C:\Users\Caleb\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\tracks.py", line 169, in extrude_holes
    drl = Draft.makeShape2DView(holes, FreeCAD.Vector(0.0, 0.0, 1.0))
  File "C:\Program Files\FreeCAD 0.20\Mod\Draft\draftmake\make_shape2dview.py", line 62, in make_shape2dview
    obj.Base = baseobj

type must be 'DocumentObject' or 'NoneType', not list
```

If I comment out all hole-related lines in [this](https://github.com/easyw/kicadStepUpMod/blob/master/tracks.py#L409-L426) section of the code, it imports the tracks & fills successfully. The Kicad file that causes this issue is available below.

[etl_lgad_mockup_v1p1.zip](https://github.com/easyw/kicadStepUpMod/files/9202931/etl_lgad_mockup_v1p1.zip)

Kicad Version: 6.0
FreeCAD Version: 0.20
kicad Stepup Version: v10.14.0

easyw, 2022-07-28:
@cfangmeier 
it is a marginal user case (no holes), but your report was so well done that it deserved attention.
it should be now fixed https://github.com/easyw/kicadStepUpMod/commit/2d46ee911e2dcc05b10025380ec1a82c75b07af4

cfangmeier, 2022-07-28:
That fixed it! Thanks for the quick response and for making this great tool!

closed 2022-07-28 by cfangmeier