easyw/kicadStepUpMod #144: FreeCAD 0.20.29177 (Git), design KiCAD Version: (6.0.7), release build

RolandWa, 2022-08-12:
Crash FreeCAD 👍 

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * dxf-library
  * EM
  * fasteners 0.4.1
  * fcgear 1.0.0
  * kicadStepUpMod 10.14.5
[/code]

  Kicad 

Application: KiCad (64-bit)

Version: (6.0.7), release build

Libraries:
	wxWidgets 3.1.7
	libcurl/7.83.1-DEV Schannel zlib/1.2.12

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jul 26 2022 02:49:38
	wxWidgets: 3.1.7 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.83.1-DEV
	ngspice: 37
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

Load design show error :

13:04:47  Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "C:\Users\rwache\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 928, in Activated
    kicadStepUptools.onLoadBoard()
  File "C:\Users\rwache\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUptools.py", line 6607, in onLoadBoard
    modules,nsk = DrawPCB(mypcb,SketchLayer,override_pcb,keep_pcb_sketch)
  File "C:\Users\rwache\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUptools.py", line 13238, in DrawPCB
    doc.getObject(board_name).ViewObject.dropObject(doc.getObject(boardG_name),App.DocumentObject,'',[])

kicadstepup, 2022-08-12:
hi, would you mind to post a zipped file which is causing the issue?
you can post me the file i.e. at kicad.info forum (maui there)

easyw, 2022-08-12:
would you mind to post your kSU settings in FC? (as image would be fine)
have you configured kSU to use 'Assembly3 Link Groups'?
it is a long time I don't test this option...

RolandWa, 2022-08-26:
Hi easyw,

No problem in the attachment you could find project file

Roland

RolandWa, 2022-08-26:
[autoprobe.zip](https://github.com/easyw/kicadStepUpMod/files/9432130/autoprobe.zip)

easyw, 2022-08-29:
@RolandWa 
the issue is related to the Assembly3 LinkGroups setting in kSU preferences.
I'm going to patch it...

ATM you could uncheck it to bypass it:
[image]

RolandWa, 2022-08-29:
I'm impressed this defect was fixed now.
Thanks easyw
Please close this bug

closed 2022-08-29 by RolandWa