easyw/kicadStepUpMod #145: missing modules - cant find KISYS3DMOD

SpeedrunnerG55, 2022-08-18:
I configured the 3d folder to use the folder that is in the kicad paths, and yet it can not find any of the 3d modules

[image]

when I try to import my PCB

[image]

[image]

I am using freecad 0.20.29177

```
OS: Ubuntu Core 20 (ubuntu-communitheme:ubuntu:GNOME/ubuntu-communitheme-xorg)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) Snap 161
Build type: Release
Branch: tag: 0.20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.14.5
```

with KiCad 

```
Application: KiCad 6.0.7

Version: 6.0.7-f9a2dced07~116~ubuntu20.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.4.0-124-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu-communitheme-xorg, x11

Build Info:
	Date: Jul 26 2022 19:32:41
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.84.0
	ngspice: 36
	Compiler: GCC 9.4.0 with C++ ABI 1013

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

easyw, 2022-08-19:
@SpeedrunnerG55 
are you using FreeCAD as snap package?
You may be interested in this issue https://github.com/easyw/kicadStepUpMod/issues/140
If you are on Snap, it seems an issue related to Snap sandbox

> Snap is sandboxed and usually can’t write to certain areas of the filesystem.

Here a similar discussion for KiCAD Snap
https://forum.kicad.info/t/kicad6-0-doesnt-show-my-linux-folders-solved/37164

May be we should report FreeCAD Snap maintainers that FC Snap is not able even to read in some area of the filesystem

SpeedrunnerG55, 2022-08-19:
[image]
I believe im using the Snap install and not the flatpack
[image]
I just installed freecad using the ubuntu software center. should I reinstall with apt?

easyw, 2022-08-19:
@SpeedrunnerG55 
you could:
- uninstall snap
- use [FreeCAD AppImage](https://github.com/FreeCAD/FreeCAD/releases/download/0.20/FreeCAD-0.20.0-Linux-x86_64.AppImage) (as it is the suggested method by the developers)
or
- keep snap 
- use sudo mount bind to bind the `/usr/share/kicad/3dmodels` to a folder in your home path
or
-  uninstall snap
- install FreeCAD using `sudo apt install freecad`

Still there is an issue in FC Snap package that the packagers should fix...

luzpaz, 2022-08-20:
Please open an issue for in the [FreeCAD-snap](https://github.com/FreeCAD/FreeCAD-snap) repo

easyw, 2022-08-29:
@SpeedrunnerG55 @luzpaz 
I'm going to add some tips on Snap and Flatpackage on how to workaround this issue

closed 2022-10-17 by easyw

Knochi, 2023-09-20:
> @SpeedrunnerG55 @luzpaz I'm going to add some tips on Snap and Flatpackage on how to workaround this issue using the '`bind`' option

Can i find this tips somewhere?