easyw/kicadStepUpMod #146: Load Track error

RolandWa, 2022-08-29:
FreeCAD 

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * dxf-library
  * EM
  * fasteners 0.4.3
  * fcgear 1.0.0
  * kicadStepUpMod 10.14.6
[/code]

Application: KiCad (64-bit)

Version: (6.0.7), release build

Libraries:
wxWidgets 3.1.7
libcurl/7.83.1-DEV Schannel zlib/1.2.12

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
Date: Jul 26 2022 02:49:38
wxWidgets: 3.1.7 (wchar_t,wx containers)
Boost: 1.79.0
OCC: 7.6.0
Curl: 7.83.1-DEV
ngspice: 37
Compiler: Visual C++ 1929 without C++ ABI

Build settings:
KICAD_USE_OCC=ON
KICAD_SPICE=ON

Load design OK,
Load Track by KSU Tools Add Track show error :

16:28:57  Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
  File "C:\Users\rwache\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 3778, in Activated
    add_toberemoved = tracks.addtracks()
  File "C:\Users\rwache\AppData\Roaming\FreeCAD\Mod\kicadStepUpMod\.\tracks.py", line 485, in addtracks
    FreeCAD.ActiveDocument.getObject('Board_Geoms'+ftname_sfx).ViewObject.dropObject(topPads,None,'',[])

argument 2 must be App.DocumentObject, not None

Orginal desing below

[autoprobe.zip](https://github.com/easyw/kicadStepUpMod/files/9445390/autoprobe.zip)

easyw, 2022-08-29:
@RolandWa 
it should now be fixed
https://github.com/easyw/kicadStepUpMod/commit/eb49b4970a8f1c9539e7d960093c77f1f40035c2

RolandWa, 2022-08-30:
Hi easyw,

New version resolve issue, please close the defect.

closed 2022-08-30 by RolandWa