easyw/kicadStepUpMod #148: Huge VWRL file size

maltaisn, 2022-09-13:
I'm exporting a part from FreeCAD in VWRL and STEP format. The wrl file takes up 30 MB and the step file is only 192 kB. The part is a 153-pin 0.5mm pitch BGA package commonly used for eMMC. Most of the file size is taken up by description of BGA solder balls: it takes 240 kB to describe a single solder ball. I expect this is not normal as solder balls are simply truncated spheres, so perhaps I've done something wrong. 
- I exported the file via the "Export 3D model to KiCAD" button,
- Chose 0.03 mesh deviation (larger values only slightly better, e.g. 27MB if 10 used)
- 0.5 crease angle (other values not better)
- Material selection left as default.

Here's a reduced version of the part with only the 4 corner balls. The wrl file still takes 1.0MB with only that.
[BGA-153_11.5x13.0mm_P0.5mm.zip](https://github.com/easyw/kicadStepUpMod/files/9552731/BGA-153_11.5x13.0mm_P0.5mm.zip)

- FreeCAD 0.19
- Latest Kicad StepUp plugin version.
- I can provide settings files as needed.

What am I doing wrong?

easyw, 2022-09-13:
> 153-pin 0.5mm pitch BGA package...
...
it takes 240 kB to describe a single solder ball

wrl is a tessellated format... to describe a sphere you need to add a lot of triangles... the better quality you want, the larger size you get;
step is a geometric format... the sphere there is a single geometrical object 

I would suggest to create a BGA model with only the external balls to reduce the wrl model

this is the way the kicad library is building its 3d models
https://kicad.github.io/packages3d/Package_BGA
i.e. BGA-153_8.0x8.0mm_Layout15x15_P0.5mm_Ball0.3mm_Pad0.25mm_NSMD.step

maltaisn, 2022-09-13:
Is there a way to reduce the number of triangles used? I'd take the suggestion of only putting the external balls but the file is still over 7 MB that way. Files in the KiCAD library are much less than that.

easyw, 2022-09-13:
you may consider to generate the model with the scripts and then make a PR to add it to the repo
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator
https://github.com/easyw/kicad-3d-models-in-freecad/blob/master/cadquery/FCAD_script_generator/BGA_packages/cq_parameters.py
or when you export the model with kSU, simply change the suggested mesh deviation value to i.e. 0.1 instead of 0.03 (default)
it should reduce a lot your wrl file size

closed 2022-09-15 by easyw

maltaisn, 2022-09-15:
Alright, got it under 1 MB with all the outer pins. Thanks.