easyw/kicadStepUpMod #150: Copper Text Missing

EderNavarro, 2022-09-22:
Hi, first of all, I want to acknowledge the great work been done here. having the possibility to join kicad and freecad is great. 

I'm triying to mill the pcb using freecad, but when I import the copper the text I put in the copper doesn't appear. the images converted to logos works

[image]

Maybe I'm doing something wrong, Hope you can help me.

Thanks.

easyw, 2022-09-23:
@EderNavarro 
for milling the pcb you should have a look at [fcad_pcb](https://github.com/realthunder/fcad_pcb) which is the library I use in my pcb to create the tracks.
You may open an issue there...

> having the possibility to join kicad and freecad is great.

thanks for the feedback

luzpaz, 2023-11-02:
@EdrZero can  you open a ticket on https://github.com/realthunder/fcad_pcb/issues