easyw/kicadStepUpMod #152: zone.min_thickness used at the wrong place

juddidier, 2022-10-07:
In file <kicad_parser.py> on line #1914 (function makeZones) remove the code-sequence <+ z.min_thickness*0.5>.

To see the erroneous effect, set in KiCAD "Minimum with" of a copper zone to eg. 0.5mm. Import the .kicad_pcb in FreeCad (with AddTracks) and you get overbloated copper zones.
The comment on Lines #1908-1912 do not apply to my version of KiCAD (6.0.6). It might be that it once was an issue, but doesn't apply anymore.

easyw, 2022-10-08:
@juddidier 
could you please add a failing kicad_pcb file? That would help in testing the issue.

> The comment on [Lines #1908-1912](https://github.com/easyw/kicadStepUpMod/blob/01930e6710b19f7a864736f8ea4edd448f48990b/kicad_parser.py#L1908-L1912)

this parser is a library I include in my wb...
Could you add an issue also at the original repository? [fcad_pcb](https://github.com/realthunder/fcad_pcb)

easyw, 2022-10-14:
@juddidier 
it should be fixed now... would you mind to check it?

juddidier, 2022-10-17:
Hi easyw,

You are almost there. To my oppinion the "min_thickness" or (newly) "thickness" has nothing to do here. I have found that in the other repo issue 59 covers this.

in KiCAD the copper zone aligns exactly with the pad:
[image]

in FreeCAD instead the copper zone gets 0.0127mm bigger:
[image]