easyw/kicadStepUpMod #156: Specify models to exclude from blacklist

jamesfowkes, 2022-12-12:
The blacklist feature allows users to exclude components from the generated model from on the basis of name, minimum height or minimum volume.

There may be certain components that would be blacklisted, but users wish to include in the model. An example is indicator LEDs, which may fall below specified size limits, but need to be included because they line up with casing cutouts or light pipes.

I suggest allowing a method of specifying these components so they they remain in the model. Since this is PCB specific, I don't think it can be in the global kicadStepUpMod configuration. It would have to be specified in the component properties or an associated file.

easyw, 2022-12-26:
this would be to add a whitelist ... it may be useful in some user case

jamesfowkes, 2022-12-26:
Yeah I'm not sure where the best place to specify that whitelist would be. 
I'd be happy do the work, I think I've found the relevant sections of code. But I'd like some guidance on where you think the list should be specified. I can think of several possible ways:

- A special component property e.g. `ksu_whitelist` that is set to `true` for those components
- A whitelist file associated with the project
- A list in a text field on a special PCB layer. (e.g on the User Comments layer)

easyw, 2022-12-27:
@jamesfowkes 
it should be fine with this commit https://github.com/easyw/kicadStepUpMod/commit/7732b422a3a7e25fa5355140c86f3126bf92ee64
I added a whitelist field with a similar policy to blacklist field (tooltip hovering the field)

closed 2022-12-30 by easyw