easyw/kicadStepUpMod #160: import silks

sqrtmo, 2023-01-26:
Hello, 
I am trying to import silkscreen dxf's and I am getting this error in the freeCAD console:
```
11:52:42  DXF libraries not found. Trying to download...
11:52:42  Running the Python command 'ksuToolsAddSilks' failed:
Traceback (most recent call last):
  File "/Users/sqrtmo/Library/Application Support/FreeCAD/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 3880, in Activated
    makefacedxf.makeFaceDXF()
  File "/Users/sqrtmo/Library/Application Support/FreeCAD/Mod/kicadStepUpMod/./makefacedxf.py", line 177, in makeFaceDXF
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/importDXF.py", line 2827, in insert
    getDXFlibs()
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/importDXF.py", line 200, in getDXFlibs
    errorDXFLib(gui)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/importDXF.py", line 118, in errorDXFLib
    progressbar = Base.ProgressIndicator()

cannot create 'N4Base19ProgressIndicatorPyE' instances
```

Mac OS ventura 13.1
ksu installed from freeCAD add-on manager and up to date
attaching photo of my import/export settings
[image]

Any suggestions?
Thanks in advance!

easyw, 2023-01-26:
@sqrtmo 
from your error log it seems you are missing the legacy DXF library
`DXF libraries not found. Trying to download...`

what do you get if you digit on python panel the following commands?
`import dxfLibrary`
`dxfLibrary.__version__`

here my results:
_'v1.40 - 2015.09.12'_

Unfortunately I had my mac broken few years ago and since there I cannot test my code on OSX anymore

sqrtmo, 2023-01-26:
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'dxfLibrary'
```

Yes, the above is my output from `import dxfLibrary`.
So it is not abled to automatically download the DXF lib?

What would be the best way to install it then?

Thanks a lot for the replay!

easyw, 2023-01-26:
@sqrtmo 
what do you get if you digit?:
```print(os.path.realpath(FreeCAD.ConfigGet("UserAppData")+"Mod"))```

sqrtmo, 2023-01-26:
../Library/Application Support/FreeCAD/Mod

easyw, 2023-01-26:
@sqrtmo 
then you should git clone in 
`../Library/Application Support/FreeCAD/Mod`
the _dxf_library_ repository found in:
https://github.com/yorikvanhavre/Draft-dxf-importer

> cd /Library/Application Support/FreeCAD/Mod
> git clone https://github.com/yorikvanhavre/Draft-dxf-importer.git dxf_importer
> 

and restart FC and test again the commands

sqrtmo, 2023-01-26:
Ok, now it is working!

Thanks a lot!

closed 2023-01-26 by sqrtmo