easyw/kicadStepUpMod #162: Track import - custom pads missing anchor shape

scottbez1, 2023-02-04:
As discussed in https://github.com/easyw/kicadStepUpMod/pull/157#issuecomment-1384443353, when tracks are imported, custom pad shapes currently only import the additional primitives, but skip the primary "`anchor`" rectangle or circle shape.

This is particularly apparent on footprints like `Infineon_PG-HSOF-8-1_ThermalVias` - see video in the linked comment above for an example.

Relevant code in kicad_parser: https://github.com/easyw/kicadStepUpMod/blob/c3d9cbf4dccf254d0a4307607bb90b06c9d9396c/kicad_parser.py#L1625-L1648

Documentation of custom pad s-expression: https://dev-docs.kicad.org/en/file-formats/sexpr-intro/#_custom_pad_options

easyw, 2023-02-05:
@scottbez1 
I've opened an issue at fcad_pcb which is the internal library used to create the pads
https://github.com/realthunder/fcad_pcb/issues/65
please feel free to add details there or make a PR

easyw, 2023-03-30:
@scottbez1 
I've added a partial solution with recent commit https://github.com/easyw/kicadStepUpMod/commit/ef6bfb0d41b52a4b6fa74c978f3404085b2ba0d1

closed 2023-04-21 by easyw