easyw/kicadStepUpMod #163: kSu 'Help Tips' tab unreadable with dark desktop theme

mmuman, 2023-02-04:
The HTML probably enforced the text color but not the background, so when using a dark theme it ends up with black text over black background, one has to select the text to be able to read it.

[image]

easyw, 2023-02-08:
@mmuman 
which is your theme andos you are using?

mmuman, 2023-02-08:
From `neofetch`:
```
OS: Debian GNU/Linux bookworm/sid x86_64 
Kernel: 6.1.0-3-amd64 
DE: Plasma 5.26.90 
WM: KWin 
WM Theme: Neon_Knights_Blue 
Theme: Materia Manjaro Dark [Plasma], Breeze [GTK3] 
Icons: Simply-Blue-Circles [Plasma], Simply-Blue-Circles [GTK2/3] 
```

easyw, 2023-02-09:
@mmuman 
it should be fixed with this commit https://github.com/easyw/kicadStepUpMod/commit/c9ca514c6bfb44ca027ee9019c2a362654604e34

would you mind to post the Help results in your OS after this update?

mmuman, 2023-02-09:
Hmm still happens with 10.7.9.
[image]

easyw, 2023-02-09:
@mmuman 
could you retry now
https://github.com/easyw/kicadStepUpMod/commit/6cc836fbdfa91b4b36d397e4a8a99c489c8d42f6

mmuman, 2023-02-09:
Almost there, just missing the second line I think.
[image]

easyw, 2023-02-09:
@mmuman 
please try it again https://github.com/easyw/kicadStepUpMod/commit/d535e9e4b532f1a8c8f84b7d43e78eb83c6391ae
would you please test the help button in kSU 

[image]

mmuman, 2023-02-09:
Everything seems fine, thanks! :+1: 
[image]

closed 2023-02-09 by mmuman