easyw/kicadStepUpMod #164: footprint don't load layers

jneiva08, 2023-03-15:
Hi,

I'm using kicadStepUpMod v10.17.0 with FreeCAD c0.20.2. My KiCad version is v7.0.1 and OS is Linux Mint 21.1

When try to load a footprint it only shows the pad's, don't load F.Fab or F.Silkscreen, as you can see in the screenshot.
The footprint i used is the following:  [Battery_THT_18650.kicad_mod.txt](https://github.com/easyw/kicadStepUpMod/files/10985371/Battery_THT_18650.kicad_mod.txt)

Screenshot:

[image]

easyw, 2023-03-16:
@jneiva08 
thanks for giving all the infos...
ATM StepUp doesn't parse correctly kv7 footprints for drawing layers ...
I'm working on an alternative parser for footprint loader... I will update it shortly

jneiva08, 2023-03-16:
Thanks for explaining.

I'm trying to convert to two step to make a PAD and its giving error.
Is better to open a new issue? or can post here?

easyw, 2023-03-16:
> I'm trying to convert to two step to make a PAD and its giving error.
> Is better to open a new issue? or can post here?

please open a different issue

closed 2023-03-18 by jneiva08