easyw/kicadStepUpMod #165: Step to footprint

jneiva08, 2023-03-16:
Hi,

I'm trying to convert the following step files to generate a PAD each. Step files: [step.zip](https://github.com/easyw/kicadStepUpMod/files/10992469/step.zip)

I'm following this guide to try to generate https://www.youtube.com/watch?v=5IcAyfhaf2g&t=1s

When need rename the sketch I used "Pads_Poly", or should use other? FreeCAD project: [freecad.zip](https://github.com/easyw/kicadStepUpMod/files/10992653/freecad.zip)

Error:
[image]

System:
I'm using kicadStepUpMod v10.17.0 with FreeCAD c0.20.2 . My KiCad version is v7.0.1 
OS is Linux Mint 21.1, python: 3.8.10

easyw, 2023-03-16:
Hi,
for polylines you need to add a circle inside the poly as a reference for the polyline pad
There are samples in the kSU Demo Menu 
[image]

I'm attaching a sample:
[pads-v2.FCStd.zip](https://github.com/easyw/kicadStepUpMod/files/10994608/pads-v2.FCStd.zip)

jneiva08, 2023-03-17:
Sorry for continuing to bother you.
Have make a new from zero to try to understand how it works. 
Have added the circle and now can create the footprint file, but when open only have the `Value` and `REF**`.
I noticed that the sketch has the dimensions.

[pad1.FCStd.zip](https://github.com/easyw/kicadStepUpMod/files/11002511/pad1.FCStd.zip)

Thanks

easyw, 2023-03-18:
@jneiva08 
no problem... your approach to build yourself is correct.
I just tried fast your file and I found your same issue.
I'll be back in short time, when I'll be in front at my pc..

easyw, 2023-03-18:
@jneiva08 
please find attached the file fixed
[pad1-fixed.FCStd.zip](https://github.com/easyw/kicadStepUpMod/files/11009593/pad1-fixed.FCStd.zip)
I have:
1) rotated the pad to have it aligned to top view
2) created a 2dview projection (kSU button)
3) converted from the 2dview a sketch (kSU button)
4) added a circle as reference for polyline pads
5) exported the fp selecting the 'pad-fp' container

What could have failed I think was creating the projection non having the pad on top view.

jneiva08, 2023-03-19:
@easyw 
Have made the conversion with success.
As did you mention, the problem is when import the step file, wasn't in the top view.
Needed to rotate it to be in top view. Then did all the steps and worked.

All this work was to make the center PAD for ORG1510 GNSS footprint.

Recommended footprint:
[image]

KiCad footprint:
[image]

easyw, 2023-03-20:
@jneiva08 
thanks for confirming you have solved. 
Nice seeing the plugin is used to solve complex footprint design; that was the initial need I had to create this part of the tool.