easyw/kicadStepUpMod #167: Preferences is squished.

mwoehlke-kitware, 2023-04-04:
Many of the labels are unreadable or nearly so:

[image]

Possibly related to display scaling?

easyw, 2023-04-04:
@mwoehlke-kitware 
would you mind to post your full FreeCAd info?
Moreover from your picture it seems the preferences are squished in all preferences tabs/wbs...
Probably is an issue in FreeCAD or your video drivers and your display scaling or your FC stylesheet (i.e. Dark Theme font size)
PS I'm using FC on a High DPI monitor without issues
[code]
OS: Windows 10 Version 1909
Word size of FreeCAD: 64-bit
Version: 0.21.0.32637 (Git)
Build type: Release
Branch: master
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
[/code]

mwoehlke-kitware, 2023-04-04:
The side icon menu does also seem broken. However, native preferences pages seem (mostly) okay. For example:
[image]

(The color picker button might be too small, but the rest looks reasonable. I do notice the odd widget here and there that seems dodgy, but for the most part, native stuff isn't nearly the mess that KSU's page is. I don't know if that's KSU's fault specifically, or due to the way add-on preference pages are implemented.)

I don't know what you mean by "full FreeCAD info"? I am using:
```
freecad-0.20.2-1.fc37.x86_64
qt5-qtbase-5.15.8-6.fc37.x86_64
python3-3.11.2-1.fc37.x86_64
```

I doubt very much it's a video driver issue; it's almost certainly due to widget layout being done incorrectly. I'm not aware that I'm using a "FC stylesheet", or even how I would apply such a thing; the colors are just my regular system colors used by all applications.

The font size appears correct. It looks like the layout is being calculated according to 1x scale but it's then trying to draw everything at (correct) 2x scale within that layout.

easyw, 2023-04-04:
to get full FC info please follow as below:
> Open the Help --> About FreeCAD menu then click on the "Copy to clipboard" button, and paste its content in the text input area. This will provide us valuable information on your exact version and revision number, your operating system as well as FreeCAD's internal libraries. (Mac users: the About FreeCAD menu is under the FreeCAD menu.)
Please do not attach a screen capture instead of posting the Help > About FreeCAD information from the clipboard.

Are you on Linux with a flatpack or snap version?

mwoehlke-kitware, 2023-04-04:
```
OS: Fedora Linux 37 (Thirty Seven) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.20.2.Unknown
Build type: Unknown
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.18.4
  * sheetmetal 0.2.60
```

> Are you on Linux [...]

Yes.

> [...] with a flatpack or snap version?

No. As previously noted, I'm using [`freecad-0.20.2-1.fc37.x86_64`](https://koji.fedoraproject.org/koji/buildinfo?buildID=2111363), a.k.a. the distro package.

easyw, 2023-04-05:
I tried in Mint20 with HIDPI 2x without issues...
[image]

still in your screenshot it seems kSU pref page is not enlarged as it should ...
I would need to investigate a bit more

easyw, 2023-04-10:
ATM I don't have any clue on what could be done to remove this issue.
Since your side menu is broken too, it could be a Fedora issue in handling HDPI Qt pages.

luzpaz, 2023-11-02:
This looks like an issue with FreeCAD itself and as @easyw said probably a packaging issue related to HiDPI. May be try using the AppImage ? 

v0.21.1 https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/0.21.1  
development version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds