easyw/kicadStepUpMod #168: Issue when loading a kicad_mod

leoheck, 2023-04-11:
Hi, I am trying to load a kicad_mod now, and got this.

```
21:52:08  kicad StepUp version 10.8.2
21:52:08  tolerance on vertex applied
21:52:08  KISYS3DMOD=/
module_3D_dir=//../
21:52:09  FC Version 021
21:52:09  VBO status False
21:52:09  FootPrint Loader /home/lheck/Dropbox/precisio/repos/Precisio_Station_HW/Precisio_HW_Library/Precisio_HW_Library.pretty/C_EEEFK1H221V.kicad_mod
21:52:09  opening Transaction 'opening_kicad_footprint'
21:52:09  geometry unsupported
21:52:09  Traceback (most recent call last):
21:52:09    File "/home/lheck/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 14351, in onLoadFootprint_click
21:52:09      onLoadFootprint()
21:52:09    File "/home/lheck/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 6014, in onLoadFootprint
21:52:09      routineDrawFootPrint(content,name)
21:52:09    File "/home/lheck/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 10160, in routineDrawFootPrint
21:52:09      obj=mypad
21:52:09  UnboundLocalError: local variable 'mypad' referenced before assignment
```

easyw, 2023-04-11:
@leoheck 
probably you are launching the command from the  kSU widget and not from the Toolbar... I need to fix it too... 
I have moved loading fps to a new routine... still some work to be completed...
**P.S.** attaching the C_EEEFK1H221V.kicad_mod file would help in managing the issue

leoheck, 2023-04-11:
Hi easyw, I could not share the footprint. I saw the issue while trying to improve some 3d models that were not I the right place. Then thought you would like to see the issue too. But since you have everything under control, I am going to close this.

easyw, 2023-04-11:
@leoheck 
if the footprint is so reserved you couldn't share, and because I told you that the issue is on a custom geometry, you could have shared a subset of your condidential work to test the failure...
But it seems your time is too precious to spend time in seriously giving details to investigate in finding a possible solution to your own issue.
I wonder why  you are still using my tools for your valuable work ...

leoheck, 2023-04-11:
I am not insulting. I came here to share the issue I saw. You were insulting me. 
I don't give a fuck if you share this for free or not.

leoheck, 2023-04-13:
Hey @easyw, how are you doing? Here I have a sample of that footprint. 
[TEST.zip](https://github.com/easyw/kicadStepUpMod/files/11217656/TEST.zip)

Hope it helps, have a nice day.

closed 2023-04-14 by easyw