easyw/kicadStepUpMod #171: KSU tools Add Tracks is ignoring rounded/filleted track segments

okelk, 2023-05-07:
I want to import the tracks in 3d from a kicad 7 file that has rounded tracks.
rounded segments are ignored. straight segments get imported.

easyw, 2023-05-08:
please post a minimal working example in pcb v7

easyw, 2023-05-08:
rounded tracks are loaded, rounded drawings in copper layers are ignored... 
closing it ATM
feel free to reopen when you have a MWE

closed 2023-05-08 by easyw

okelk, 2023-05-08:
I attached a MWE kicad project as a zip.
also here some screenshotz that illustrate the problem:

[image]
[image]
[stepup-rounded.zip](https://github.com/easyw/kicadStepUpMod/files/11425279/stepup-rounded.zip)

easyw, 2023-05-09:
no issue on my system... which is your kSU version?
From my FC Report panel printing:
`KiCadStepUpWB.Activated(v 10.18.8)`
`FC Version 021-0`
`kicad_parser_version 2.2.4`
`tracks version: 2.5.8`
`kicad version=20221018`

okelk, 2023-05-10:
I found the issue.

The Addon Manager told me that I'm on **Version 10.18.8**  .
...so that's what I thought I'm using

But under
Edit --> Preferences --> KicadStepupGUI
kicad StepUp version **9.7.3.9**
was shown.

So I assume there was some conflict with some older version of the plugin?

I deleted the /Mod/kicadStepUpMod.bak folder and reinstalled the plugin via the Addon Manager.
Now it works.

Thank You!