easyw/kicadStepUpMod #172: KIPRJMOD wrong path

SeriousD, 2023-05-17:
Hi,

https://github.com/easyw/kicadStepUpMod/blob/fb298571091ab2932a2f7a0953d627011d3314f4/kicadStepUptools.py#L6555
causes `name` to point to a wrong path(in the /run/$uid/...folder) and so all subsequent relative paths to be wrong.

Possible fix is to avoid using the native dialog:
```
 name, Filter = PySide.QtGui.QFileDialog.getOpenFileName(None, "Open kicad PCB File...",
                make_unicode(last_pcb_path), "*.kicad_pcb",options=QtWidgets.QFileDialog.DontUseNativeDialog)
```

Freecad info:
[code]
OS: Ubuntu Core 20 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git) Snap 517
Build type: Release
Branch: tag: 0.20.2
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.18.8
  * A2plus 0.4.60n
  * sheetmetal 0.2.63
  * ExtremeProDark 2.7.8
[/code]

easyw, 2023-05-17:
@SeriousD 
are you on FreeCAD Snap?
Do you have a failing kicad_pcb file to test?
Is your issue on 3D model loading (so related to 'KIPRJMOD') or in FreeCAD opening 'kicad_pcb' file?

SeriousD, 2023-05-17:
Yes I am on FreeCAD snap.
All kicad_pcb files with KIPRJMOD fail.
The issue occurs on loading the board into the workbench.

As stated: The filedialog returns the path to a local copy of the boardfile(/run/$uid/...) and so the path stored in "name" doesnt point to the source directory with the other files needed for the project, so the model path resolver fails.

The fix for me was to set the `options=QtWidgets.QFileDialog.DontUseNativeDialog` argument of getOpenFileName.
For my user experience this is also an improvement, since it keeps the last chosen folder.
I found this solution from [this](https://stackoverflow.com/questions/57646908/pyqt5-qfiledialog-is-not-returning-correct-paths-in-ubuntu) SO thread.

easyw, 2023-05-17:
@SeriousD 
this is not necessary for i.e. FreeCAD as AppImage...
I would try not to use native dialog only on Snap or Flat FreeCAD environment...
Could you please post your result with this python command in FC python console panel?
`FreeCAD.Version()`

SeriousD, 2023-05-17:
```
>>> FreeCAD.Version()
['0', '20', '2', '29177 +426 (Git)', 'Unknown', '2022/12/07 02:35:37', 'tag: 0.20.2', '930dd9a76203a3260b1e6256c70c1c3cad8c5cb8']
```

easyw, 2023-05-17:
@SeriousD 
thanks, it seems the Snap info is inside 'About Info()' and not in FreeCAD.Version() python command.
And what do you get with:
`import os; os.getenv('SNAP_REVISION')`

easyw, 2023-05-18:
@SeriousD 
I'm moving to add an option to select _"DontUseNativeDialog"_
[image]

easyw, 2023-05-18:
@SeriousD 
could you try it now? I've updated the configuration settings

closed 2023-05-31 by easyw

SeriousD, 2023-06-02:
works like a charm! Thanks!