easyw/kicadStepUpMod #173: translation GUI of the kicadStepUp workbench

kaktusus, 2023-06-07:
Is there even a slight chance that this cool workbench will be able to have a GUI in different languages?

https://crowdin.com/project/freecad-addons
https://github.com/FreeCAD/DevelopersHandbook/blob/master/technical/translation.md

easyw, 2023-06-08:
@kaktusus 
would you mind to have a look at this initial PR https://github.com/easyw/kicadStepUpMod/pull/142? 
I would need someone who is familiar with to help in adding a translation option...

kaktusus, 2023-06-08:
I'm just a translator ...

kaktusus, 2023-08-26:
I began to familiarize myself with the process.
Maybe in some time I will be able to help. :wink: 
Do you speak Polish?

kaktusus, 2023-08-26:
I can be the first tester and lead the pioneering translation into Polish.
But only after you merge the PR from Kuzma30.

hasecilu, 2024-03-07:
I'll take a look into this