easyw/kicadStepUpMod #175: Export 3d model  tries to write into root directory /

VK2BEA, 2023-06-17:
I've probably failed to initialize something in the preferences but recently I've been unable to export 3D (wml or step) because the 'Export 3D model' command tries to write in the root directory.
How do I define where to write the model? (It doesn't ask at the time of execution)

This is using the nightly FeeCAD 0.21.0 (KiCadStepUp 10.19.4)

```
Running the Python command 'ksuToolsExportModel' failed:
Traceback (most recent call last):
  File "/home/michael/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 1266, in Activated
    routineScaleVRML()
  File "/home/michael/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 7412, in routineScaleVRML
    lbl=go_export(fullFilePathName)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/michael/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 2684, in go_export
    export(objs, fPathName, 0.3937, lbl)
  File "/home/michael/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 2345, in export
    exportVRMLmaterials(meshes, filename)
  File "/home/michael/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 2041, in exportVRMLmaterials
    with builtin.open(fpath, 'w') as f: #py3
         ^^^^^^^^^^^^^^^^^^^^^^^^

[Errno 13] Permission denied: '/Molex-1625.IC260A001.wrl'
```

VK2BEA, 2023-06-17:
OK, it seems like there is a subtle bug...
If FreeCAD is started from the command line with the file as an argument ...
`$ FreeCAD Molex-1625.IC260A.FCStd`
this problem occurs.
If FreeCAD is started without an argument and then the file is opened explicitly when in FreeCAD, the 3D models are saved in the same directory as the FCStd file.

easyw, 2023-06-17:
@VK2BEA 
thank you for your detailed feedback...
it seems that the document 'filename' includes the full path only if it is opened by menu and not using the command line argument...
As a workaround you can save the file before exporting... this operation will add the full path to document filename
`App.ActiveDocument.FileName`

VK2BEA, 2023-06-17:
At least for an existing file, opening and selecting 'save' did not work around this problem; however, using 'save as' and overwriting the same file name did.

easyw, 2023-06-18:
@VK2BEA 
an other option would be to launch the command using the full path for the file i.e.:
`$ FreeCAD  ~/Documents/Molex-1625.IC260A.FCStd`