easyw/kicadStepUpMod #176: Provide support for User.x layers

transistorgrab, 2023-06-25:
Hi,

I have a rather complex design and I opted to put mechanical outlines meant for export to DXF/FreeCAD in the User.3 layer to prevent possible collisions from imported footprints.

Someone mentioned KicadStepUp as a tool for this to prevent the extra step of DXF ex-/import.
However, none of the "User." layers are available as an import option and I could not find preferences option to add them to the list.

I have no idea if I'm the only one using these layers but I would love having them as an option.

transistorgrab, 2023-06-29:
Hi @easyw,
as described I use the User.3 layer.
I saw the available layers but I would need to change all affected footprints to map it to some of the available layers. 
And because some footprints out there use some of the "old" layers for various things I used User.3 to prevent any accidental items showing up in my "mechanical layer".
Therefore, I currently use DXF export and import in FreeCAD.

I would prefer if KicadStepUp would provide all Kicad layers to choose from. 
I don't know if it would generate much effort for the project but to me it looks like an arbitrary selection based on your(?) use cases which in my case prevent me to use it without extra work/changes in my design.
(My design came first. I learned only much later about KicadStepUp and its usefulness).

Is it that Kicad introduced these layers only recently and therefore, StepUp did not include them (yet)?

I also think it would be helpful for new users if in KicadStepUp the layers had the same name as in Kicad. i.e. "*.User" in KicadStepUp are "User.*" in Kicad. 

I only learned by trying that in StepUp the "*.User" are the same as "User.*". This was a nice surprise but an unexpected result with my experience with IT products. :)

easyw, 2023-06-29:
> I also think it would be helpful for new users if in KicadStepUp the layers had the same name as in Kicad. i.e. "*.User" in KicadStepUp are "User.*" in Kicad.

these are the real internal format layer names (open kicad_pcb file in text editor):
```
    (0 "F.Cu" signal)
    (31 "B.Cu" signal)
    (32 "B.Adhes" user "B.Adhesive")
    (33 "F.Adhes" user "F.Adhesive")
    (34 "B.Paste" user)
    (35 "F.Paste" user)
    (36 "B.SilkS" user "B.Silkscreen")
    (37 "F.SilkS" user "F.Silkscreen")
    (38 "B.Mask" user)
    (39 "F.Mask" user)
    (40 "Dwgs.User" user "User.Drawings")
    (41 "Cmts.User" user "User.Comments")
    (42 "Eco1.User" user "User.Eco1")
    (43 "Eco2.User" user "User.Eco2")
    (44 "Edge.Cuts" user)
    (45 "Margin" user)
    (46 "B.CrtYd" user "B.Courtyard")
    (47 "F.CrtYd" user "F.Courtyard")
    (48 "B.Fab" user)
    (49 "F.Fab" user)
    (50 "User.1" user)
    (51 "User.2" user)
    (52 "User.3" user)
    (53 "User.4" user)
    (54 "User.5" user)
    (55 "User.6" user)
    (56 "User.7" user)
    (57 "User.8" user)
    (58 "User.9" user)
```

> I only learned by trying that in StepUp the ".User" are the same as "User.". This was a nice surprise but an unexpected result with my experience with IT products. :)

I had an unexpected result with my experience with IT products when kicad dev changes exposed names compared to internal format names 🤣 

> I just saw that StepUp has layers like "FillZone", "KeepOutZone". I cannot find these in Kicad (7). And I would not know if this maps on any Kicad layer which one this would be.

> I only learned by trying that in StepUp ...

unfortunately the documentation is lacking (any help would be appreciated)...
Anyway opening the cheatsheet pdf you could see there is a Demo Menu in the StepUp wb... there you can find footprint examples that would try to explain the use of these extra layers

easyw, 2023-09-22:
@transistorgrab 
now it is possible to handle almost all layers

closed 2023-09-26 by easyw