easyw/kicadStepUpMod #179: FreeCAD.ActiveDocument.getObject('Board_Geoms'+tname_sfx).addObject(extrude) - has no attribute addObject

RolandWa, 2023-08-18:
FreeCAD 
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Alternate_OpenSCAD 1.0.0
  * dxf-library
  * FreeCAD-OpenEMS-Export
  * FreeCAD-OpenEMS-Export.7z
  * kicadStepUpMod 10.19.4

KiCAD
Application: KiCad Schematic Editor x64 on x64

Version: 7.0.7, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Aug 14 2023 02:42:39
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.7.1
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

During import track error :

10:40:50  running time: 4.148sec
10:40:50  TBD: gr_rect,gr_poly use stroke width (w makethickline)
10:40:50  TBD: gr_rect,gr_poly use stroke width (w makethickline), FC build date
10:40:50  making zones...
10:40:50    making 2 polygons...
10:40:50    region 1/2, holes: 2
10:40:50  region 2/2, holes: 40
10:40:51  polygons done
10:40:51  fetch holes 'holes_wire' from cache
10:40:55  zones done
10:40:55  running time: 8.673sec
10:40:55  running time: 8.784sec
10:40:55  Sketch001
10:40:55  Running the Python command 'ksuToolsAddTracks' failed:
Traceback (most recent call last):
  File "\FreeCAD\Mod\kicadStepUpMod\.\kicadStepUpCMD.py", line 4159, in Activated
    add_toberemoved = tracks.addtracks()
  File "\FreeCAD\Mod\kicadStepUpMod\.\tracks.py", line 628, in addtracks
    topPads_cut_Name, temp_tobedeleted = cut_out_tracks(pcb_sk,topPads,ftname_sfx)
  File "\FreeCAD\Mod\kicadStepUpMod\.\tracks.py", line 238, in cut_out_tracks
    FreeCAD.ActiveDocument.getObject('Board_Geoms'+tname_sfx).addObject(extrude)

'App.DocumentObject' object has no attribute 'addObject'

easyw, 2023-08-27:
@RolandWa 
do you have a kicad_pcb file to test?
without a failing minimal sample it is not possible to check your issue

RolandWa, 2023-08-28:
[noise_source.zip](https://github.com/easyw/kicadStepUpMod/files/12451106/noise_source.zip)
Hire you are, the project file

easyw, 2023-09-01:
Hi,
I cannot reproduce your issue...
[noise_source-FC.zip](https://github.com/easyw/kicadStepUpMod/files/12495609/noise_source-FC.zip)

easyw, 2023-11-13:
it should now solved with recent update

closed 2023-11-13 by easyw