easyw/kicadStepUpMod #182: unnamed hole import problem

onazir, 2023-09-13:
Hi,

I am trying to open a footprint file in freecad but unnamed holes positions are disordered. 

report view

14:59:29  PoM not present
14:59:29  using 'Part' container and 'Links'
14:59:29  FC Version 021-1
14:59:29  Asm3 WB present
14:59:29  kicad StepUp version 10.8.7
14:59:29  tolerance on vertex applied
14:59:29  applying Materials to Shapes
14:59:29  your home path is C:\Users\nazir
14:59:29  export to STEP False
14:59:29  Tabified done !
14:59:29  done!
14:59:29  kicad StepUp version 10.8.7
14:59:29  tolerance on vertex applied
14:59:29  PoM not present
14:59:29  using 'Part' container and 'Links'
14:59:29  FC Version 021-1
14:59:29  Asm3 WB present
14:59:29  kicad StepUp version 10.8.7
14:59:29  tolerance on vertex applied
14:59:29  applying Materials to Shapes
14:59:29  your home path is C:\Users\nazir
14:59:29  kicadStepUp
14:59:29  closed
14:59:29  fp loader v1.1.0
14:59:29  FC Version 021-1
14:59:29  kicad_parser_version 2.2.9
14:59:29  footprints version: 1.1.0
14:59:29  G:/My Drive/projeler/CustomParts/Footprints/Modules.pretty
[nvidia jetson orin nano module.zip](https://github.com/easyw/kicadStepUpMod/files/12597157/nvidia.jetson.orin.nano.module.zip)
/nvidia jetson orin nano module.kicad_mod
14:59:29  cannot find kicad_common
14:59:29  kicad version=20211014
14:59:29  kicad v5 stack missing
14:59:29  file removed
14:59:29  making holes
14:59:29  making holes...
14:59:29    pad holes: 4, skipped: 0
14:59:29    oval holes: 0
14:59:29    via holes: 0, skipped: 0
14:59:29    total holes added: 4
14:59:29  holes done
14:59:29  making Top Pads
14:59:29  making pads...
14:59:29    checking edge cuts
14:59:29      making 12 lines
14:59:29      making 2 circles
14:59:29      making 1 rects
14:59:29    making 12 lines
14:59:29    making 2 circles
14:59:29    making 1 rects
14:59:29    footprints: 1
14:59:29    pads: 266, skipped: 2
14:59:29    vias: 0, skipped: 0
14:59:29    total pads added: 264
14:59:29  pads done
14:59:39  making Bot Pads
14:59:39  making pads...
14:59:39    checking edge cuts
14:59:39      making 12 lines
14:59:39      making 2 circles
14:59:39      making 1 rects
14:59:39    making 12 lines
14:59:39    making 2 circles
14:59:39    making 1 rects
14:59:39    footprints: 1
14:59:39    pads: 266, skipped: 264
14:59:39    vias: 0, skipped: 0
14:59:39    total pads added: 2
14:59:39  pads done
14:59:39  <Part> ViewProviderExt.cpp(1267): Cannot compute Inventor representation for the shape of Unnamed#Shape002: Bnd_Box is void
14:59:39  <Part> ViewProviderExt.cpp(1267): Cannot compute Inventor representation for the shape of Unnamed#Shape002: Bnd_Box is void
14:59:39  making Top Net Ties
14:59:39  making net ties...
14:59:39  Net Tie poly done
14:59:40  making Bot Net Ties
14:59:40  making net ties...
14:59:40  Net Tie poly done
14:59:40  making Drawings sketches
14:59:40  making sketches...
14:59:40  sketch done
14:59:40  making sketches...
14:59:40  sketch done
14:59:41  making sketches...
14:59:41  sketch done
14:59:41  making sketches...
14:59:41  sketch done
14:59:41  making sketches...
14:59:41  sketch done
14:59:42  making sketches...
14:59:42  sketch done
14:59:42  making sketches...
14:59:42  sketch done
14:59:42  making sketches...
14:59:42  sketch done
14:59:42  making sketches...
14:59:42  sketch done
14:59:43  making sketches...
14:59:43  sketch done
14:59:43  making sketches...
14:59:43  sketch done

easyw, 2023-09-14:
@onazir 
in your footprint you have 2 NPTH pads with a pad size biggert than drill hole...
this is not supported ATM... 
This setting in real production would produce two THP pads and not two NPTH pads.
(PADs with Diameter 6.0mm Hole 3.7mm)

onazir, 2023-09-15:
[jetson nano footprint.pdf](https://github.com/easyw/kicadStepUpMod/files/12616223/jetson.nano.footprint.pdf)

thank you for respond 

I will make it change and try again.

onazir, 2023-09-15:
I change the mechanicals to th and same result.
[image]

09:21:04  cannot find kicad_common
09:21:04  kicad version=20211014
09:21:04  kicad v5 stack missing

easyw, 2023-09-18:
@onazir 
these messages are just warning
````
09:21:04 cannot find kicad_common
09:21:04 kicad version=20211014
09:21:04 kicad v5 stack missing
````
the issue is related to the rotation of the 3D mode, not correctly handled ATM
if you remove the rotation in the kicad_mod file, FC will be loaded correctly

[nvidia jetson orin nano module-rev1-FC-kicad_mod.zip](https://github.com/easyw/kicadStepUpMod/files/12652296/nvidia.jetson.orin.nano.module-rev1-FC-kicad_mod.zip)

[image]

I'm going to check the code to see if I can fix it.

closed 2023-09-19 by onazir

easyw, 2023-09-22:
this should be solved with this commit