easyw/kicadStepUpMod #183: Rotation of footprint is wrong when loaded in ksu

dennevi, 2023-09-20:
In KiCad 7.0.7 the footprint looks like this:
[image]
But when loaded into ksu 10.8.7 everything but the two holes have been rotated:
[image]
Here's the footprint with .txt added to the filename: [HM-AMP-1473005-1.kicad_mod.txt](https://github.com/easyw/kicadStepUpMod/files/12673108/HM-AMP-1473005-1.kicad_mod.txt)

I've also noticed the same thing problem with this footprint: [HM-ALUCAP-D10x10.5.kicad_mod.txt](https://github.com/easyw/kicadStepUpMod/files/12673128/HM-ALUCAP-D10x10.5.kicad_mod.txt)
In this case everything is rotated, I guess because the footprint doesn't have any holes. It has the effect that the STEP and WRL files that are exported from KSU has to be rotated in KiCad to fit the footprint.
KiCad: 
[image]
KSU: 
[image]

Thanks for a really great tool! I just started using it today.

easyw, 2023-09-21:
Hi,
the rotation is related to the 3D model settings in kicad... but probably it is better to just load the fp as in kicad and simply warn the user in case of rotations

dennevi, 2023-09-22:
> the rotation is related to the 3D model settings in kicad

Thank you! Setting the Z rotation to 0 in the footprint before importing it into FreeCad gave me the expected behavior and a much better workflow. 

> probably it is better to just load the fp as in kicad

Yes, if you ask me. I don't see why one would want the footprint to be rotated when working with 3d models for footprints. I'm not even sure that the warning is needed.

easyw, 2023-09-22:
@dennevi 
I have updated the tool... it should now read correctly the fp

dennevi, 2023-09-22:
Works really good! I've tried it with 5-6 footprints that has a rotated 3D model and a few that doesn't. Works every time and the warning only shows up when it should.

Thanks a bunch!

closed 2023-09-22 by dennevi