easyw/kicadStepUpMod #184: Model loses it's colors when it's moved

dennevi, 2023-09-21:
Some models loses all colors when I move them using the tools in KSU. The colors are lost regardless which of the buttons in KSU is used to move the model.
[image]

But the colors are not lost when the model is moved using the Align Tools in Manipulator or when using Transform.

[image]

Here's one model that this happens with: [HM-TSSOP14.FCStd.txt](https://github.com/easyw/kicadStepUpMod/files/12682972/HM-TSSOP14.FCStd.txt) (.txt added at the end to satisfy Github).

The model is coming from a step file which I imported, made a Union of, and removed all objects except the new Union. Here's the step file as well: [tssop-14.stp.txt](https://github.com/easyw/kicadStepUpMod/files/12683002/tssop-14.stp.txt)

dennevi, 2023-09-21:
Forgot to mention that I'm using kicad StepUp version 10.8.7.

I've noticed that this happens even if I don't make a Union manually. First rotation works fine, but when the model is rotated a second time it loses all color.

[image]

easyw, 2023-09-21:
Hi,
that is something related to FC versions... the behavior is changed recently... anyway I have a workaround that I need to push...
P.S. the tool for rotating and moving objects has been overtaken by my other [Manipulator WB](https://github.com/easyw/Manipulator) for FC which is more suitable for aligning models to fp.

dennevi, 2023-09-22:
> I have a workaround that I need to push

That's great! I'll test it out after you find the time to push it.

> the tool for rotating and moving objects has been overtaken by my other [Manipulator WB](https://github.com/easyw/Manipulator)

I've used that WB to align the 3d model to footprint Origin, and that's a really great tool as well! I had not noticed the Mover Tools before. I'm now trying to rotate the footprint using these tools, but I cannot figure out how to do that. The Inputs are grayed out, and no matter what I do the 3d model stays put at the footprint. The move selected control in Align tools moves the model though.

[image]

I hope that the "put on Z" button stays in KSU, or gets moved to Manipulator, because that button is awesome! 😃 
[image]

easyw, 2023-09-22:
@dennevi 
for Mover you need to enable the tool clicking on the 'cross' .. then it will be sort of 'underlined'. Then you need to click on a model you want to move (a face or an edge)...
in the page there are some useful gifs
Put on Z is available on Aligner... 
https://wiki.freecad.org/File:Manipulator-WB-Tools.png
https://wiki.freecad.org/Manipulator_Workbench

With this commit https://github.com/easyw/kicadStepUpMod/commit/03f407118dcf134c069e4f0dea6eda2315135aec the kSU tool should fix colors when using the old way

dennevi, 2023-09-22:
I've only tested it with one model, but the commit indeed seems to fix the issue. Huge thank you!

Also, I finally understand how the Mover Tools in Manipulator works! 😄 I like the way the Mover Tools work when moving in steps in X, Y, Z direction. But I still find the tools for rotating and "put to Z" in kSU more intuitive and easy to use.

dennevi, 2023-09-22:
Closing issue

closed 2023-09-22 by dennevi