easyw/kicadStepUpMod #186: not working with V0.22 (appimage)

2EyeTec, 2023-10-02:
I'm working with the weekly appimage and since some versions, KiCadStepUp is not working anymore. When I try to use it, I receive the following error (or similar, depending on the icon I click):
```
Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "/home/christian/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 1198, in Activated
    import kicadStepUptools
  File "/tmp/.mount_FreeCAksoaH3/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/christian/.local/share/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 21349, in <module>
    KSUWidget.activateWindow()

name 'KSUWidget' is not defined
```

This is my FC Version:

[code]
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34478 (Git) AppImage
Build type: Release
Branch: master
Hash: 70ca90f46f30c84f9f63d78be6cd14f5a0b93aea
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * ExplodedAssembly
  * Manipulator 1.5.0
  * LCInterlocking
  * sheetmetal 0.3.0
  * kicadStepUpMod 10.19.5
  * Assembly4 0.50.2
  * fasteners 0.4.66
[/code]

It was working with the following version:

[code]
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33679 (Git) AppImage
Build type: Release
Branch: master
Hash: 205c0e73459088308812858282ff51f09dfd2676
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * ExplodedAssembly
  * Manipulator 1.5.0
  * LCInterlocking
  * sheetmetal 0.3.0
  * kicadStepUpMod 10.19.5
  * Assembly4 0.50.2
  * fasteners 0.4.66
[/code]

easyw, 2023-10-02:
@2EyeTec 
I'm working on that
https://forum.freecad.org/viewtopic.php?p=709208#p709208
I'm preparing a patch

2EyeTec, 2023-10-02:
Oh - I did read this already incl. your messages, because I was also wondering what happened to the combo view. I did not understand that it is also causing this . I should have read the thread more carefully! Sorry & Thanks ;-) !

closed 2023-10-02 by 2EyeTec

easyw, 2023-10-11:
@2EyeTec 
with this commit https://github.com/easyw/kicadStepUpMod/commit/f8d2ff3950e407b574ea91daca5cd3d94e72d42b
kSU should work also w/ FC0.22

2EyeTec, 2023-10-14:
Yes - it does! Thanks a lot!