easyw/kicadStepUpMod #187: Export 3D model not used new selected parameter

adamalfath, 2023-10-12:
### Step to reproduce:

Model: [SIM_Nano_GCT_SIM8060-6-x-14-00-A.zip](https://github.com/easyw/kicadStepUpMod/files/12884971/SIM_Nano_GCT_SIM8060-6-x-14-00-A.zip), or could be any model

**_Single export operation:_**
1. Open file
2. Export, MeshDeviation = 0.01, CreaseAngle = 0.5, Color = as is
3. WRL file size = 13372kB
4. Close FreeCAD, open file again
5. Export, MeshDeviation = 0.1, CreaseAngle = 0.5, Color = as is
6. WRL file size = 669kB

**_Multiple export operation:_**
1. Open file
2. Export, MeshDeviation = 0.01, CreaseAngle = 0.5, Color = as is
3. WRL file size = 13372kB
4. Export again use new parameter, MeshDeviation = 0.1, CreaseAngle = 0.5, Color = as is
6. WRL file size = 13372kB (not changed)

Another observed behavior is that when the previous MeshDeviation value is larger than the later, it seems exported correctly. But if it's the other way around, it's not. For example (arbitrary value):
- #1MD=0.1, WRL=1kB -> #2MD=0.01, WRL=10kB -> #3MD=0.001, WRL=100kB (expected result, smaller MD = larger file size)
- #1MD=0.001, WRL=100kB -> #2MD=0.01, WRL=100kB -> #3MD=0.1, WRL=100kB

### Expected
The exported model should use a newly selected parameter

### Version
FreeCAD 0.20.2
KSU 10.8.7

easyw, 2023-10-14:
I have just had a look at the code and it seems there is a sort of caching for meshing method in FC.
If this is confirmed, I'm not sure it could be solved in kSU

adamalfath, 2023-10-14:
I found the same issue https://github.com/FreeCAD/FreeCAD/issues/7393 and its forum thread https://forum.freecad.org/viewtopic.php?t=71391. I tried to implement the workaround mentioned there and it seems to follow the expected result (correct filesize vs MeshDeviation).

```
def shapeToMesh(shape, color, transp, mesh_deviation, scale=None):
    #mesh_deviation=0.1 #the smaller the best quality, 1 coarse
    #say(mesh_deviation)
    mesh_data = shape.tessellate(mesh_deviation)

    # New mesh method
    mesh = MeshPart.meshFromShape(Shape = shape,
                                  LinearDeflection = mesh_deviation, # Don't know if this is the same
                                  AngularDeflection = radians(10)) # ?
    # Update mesh_data
    mesh_data_list = list(mesh_data)
    mesh_data_list[0] = [point.Vector for point in mesh.Points]
    mesh_data_list[1] = [facet.PointIndices for facet in mesh.Facets]
    mesh_data = tuple(mesh_data_list)

    points = mesh_data[0]
    if scale is not None:
        points = map(lambda p: p*scale, points)
    newMesh= Mesh(points = points,
                faces = mesh_data[1],
                color = color, transp=transp)
    return newMesh
```
Also found this https://forum.freecad.org/viewtopic.php?t=32889, mention meshing all shape before the individual faces. I insert `shape1.tessellate(mesh_dev, True)` before this [line](https://github.com/easyw/kicadStepUpMod/blob/1d6abe06b2e7da5adf48292d7d7ffb8b55eb6a3d/kicadStepUptools.py#L2337). It also seems behaving correctly.

**_Results:_**
|Parameter|Original|meshFromShape|shape1.tessellate|
|-|-|-|-|
|MD=0.1, CA=0.5|669kB|1693kB|2684kB (why it's higher?)|
|MD=0.01, CA=0.5|13372kB|1859kB|13372kB (same)|

Disclaimer: I'm a novice regarding Python, let alone FC/KSU sources. I just followed the mentioned workaround and don't know what is the difference between each method and its implication. Maybe the ultimate "fix" is just to find where this FC mesh caching is live and delete it before creating the new mesh.

easyw, 2023-10-15:
@adamalfath 
thanks for your detailed feedback! 
I'm going to include your code snips.

easyw, 2023-10-16:
@adamalfath 
could you please check if it is solved with this commit https://github.com/easyw/kicadStepUpMod/commit/40cd37c5f54b6c34fb90848960f2e23407ac4d2e?

adamalfath, 2023-10-16:
@easyw 
Yes, the output file now seems correlated to the updated parameter. Thanks.
Although, I might be over-complicated things. My bad. 

I found this old issue tracker https://tracker.freecad.org/view.php?id=1269, so basically we only need to set the second argument for this [tesselate function](https://github.com/easyw/kicadStepUpMod/blob/1d6abe06b2e7da5adf48292d7d7ffb8b55eb6a3d/kicadStepUptools.py#L2020C49) to true: `shape.tessellate(mesh_deviation, True)`. No need to mesh the whole shape or use the new meshing method. That will clear the old mesh (iterated for all faces = all calculated mesh) and solve the initial problem. 😅

I've done several tests and this works too.

easyw, 2023-10-16:
@adamalfath 
thanks again for your contribution...
I refined the code adding just the code `shape.tessellate(mesh_deviation, True)`
https://github.com/easyw/kicadStepUpMod/commit/b1a2a6bea5bc7be4fa6c71f821d1ffd71104a17d
could you please check it again?

closed 2023-10-20 by easyw

adamalfath, 2023-10-30:
@easyw 
I am so sorry for the late response, caught up on rl job.
I pulled the mentioned commit version and it solved my initial problem.
Thank you very much for the support.