easyw/kicadStepUpMod #188: Memory leak importing silkscreens

cdntinker, 2023-10-13:
Apparently, when importing silkscreen to the model, memory is never released.
While applying the silks, memory usage continuously increases until the machine runs out.
This used memory is not even released when I close the model.
The only way I've found to get any memory back is to actually shut down FreeCAD.

Linux (LMDE5)
FreeCAD (0.19)
kicadStepUp (whatever version it updated to 2023/10/13)

easyw, 2023-10-14:
@cdntinker 
could you please check the last commit https://github.com/easyw/kicadStepUpMod/commit/1d6abe06b2e7da5adf48292d7d7ffb8b55eb6a3d ?

cdntinker, 2023-10-14:
Much better!

No more runaway memory at all.

I thank you!

easyw, 2023-10-15:
@cdntinker 
thanks for confirming. I'm closing it.

closed 2023-10-15 by easyw