easyw/kicadStepUpMod #189: local variable 'import_drawings' referenced before assignment

judge2005, 2023-10-16:
MacOS Sonoma 14.0
KiCAD 7.0.1
FreeCAD 0.20.2
KiCADStepUp 10.17.1

I try to import a .kicad_pcb file in to a new document and get this error:

opening /Users/Nemesis/Development/circuits/b13b/b13b.kicad_pcb
16:28:47  my file path /Users/Nemesis/Development/circuits/b13b
16:28:47  Running the Python command 'ksuToolsOpenBoard' failed:
Traceback (most recent call last):
  File "/Users/Nemesis/Library/Application Support/FreeCAD/Mod/kicadStepUpMod/./kicadStepUpCMD.py", line 1206, in Activated
    kicadStepUptools.onLoadBoard()
  File "/Users/Nemesis/Library/Application Support/FreeCAD/Mod/kicadStepUpMod/./kicadStepUptools.py", line 6670, in onLoadBoard
    if import_drawings and FreeCAD.ActiveDocument is not None:

local variable 'import_drawings' referenced before assignment

strooph, 2023-10-16:
I have the same issue 
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 22f524c0c755003c12234fab1bad6a9f696b661a
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: French/France (fr_FR)
Installed mods: 
  * Assembly4 0.50.4
  * Manipulator 1.5.0
  * kicadStepUpMod 10.20.6

Application: KiCad PCB Editor x86_64 on x86_64

Version: 7.0.8-7.0.8~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 6.0.0
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.16

Platform: Ubuntu 22.04.3 LTS, 64 bit, Little endian, wxGTK, ubuntu, wayland

Build Info:
	Date: Sep 30 2023 13:54:18
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.88.1
	ngspice: 38
	Compiler: GCC 11.4.0 with C++ ABI 1016

Build settings:
	KICAD_SPICE=ON

easyw, 2023-10-17:
@judge2005 @strooph 
it should now be fixed

strooph, 2023-10-18:
it works thank you!!!

best regards 
philippe

easyw, 2023-10-18:
@strooph 
thanks for confirming me the solution.
I'm closing the issue.

closed 2023-10-18 by easyw